rwaldron/johnny-five #1181: analogReference() in J5

aabm00, 2016-07-22:
There are some way n  J5  to configures the [reference voltage](https://www.arduino.cc/en/Reference/AnalogReference) used for analog input. For example work with the internal reference voltage of a board (analogReference ( INTERNAL ) to get best accuracy?

Thanks

soundanalogous, 2016-07-23:
This isn't available yet, at least not for Arduino boards. This proposal needs to be revisited and approved before the functionality will be available: https://github.com/firmata/protocol/pull/8/files.

aabm00, 2016-07-24:
thanks for the answer !!.

So it looks that isn't going be solved in a short time. Are there some prevision about it?

I am working with an arduino YUN board and the VCC can vary. It would have to be 5V but  can vary until 4.5V. So when working with sensors affects to the accuracy.
So it would be nice to have the posibility to have a stable VCC.

closed 2018-08-28 by stale[bot]