rwaldron/johnny-five #1183: how to set time interval ?

chaos39, 2016-07-24:
Hi, I would like know to how to the set interval to obtain the reading per 1 second instead of obtaining the readings faster than 1 secs. Please help me. Thanks !

This is the code that I am doing now  

var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {

  var led = new five.Led(2);
  var led1 = new five.Led(6);
  var temperature = new five.Thermometer({
    controller: "GROVE",
    pin: "A3"
  });

   var c = 0;

  temperature.on("data", function() {
    if (c === Math.round(this.celsius)) {
      return;
    }

    c = Math.round(this.celsius);
    if ( c >=30 ){
    led.on();
    led1.off();
    console.log( c + "*c" + " HOT ");
  } else if ( c <=20){
   led.off();
   led1.on();
   console.log( c + "*c" + " Cold ");
  } else {
   led.off();
   led1.off();
   console.log( c + "*c" + " Warm");
  }

  });
});

dtex, 2016-07-24:
Use the ```freq``` parameter to control how often the event is emitted. Value is in milliseconds.

Here'e the [documentation for that](http://johnny-five.io/api/thermometer/#parameters)

````
var temperature = new five.Thermometer({
  controller: "GROVE",
    pin: "A3",
    freq: 1000
});
````

Pedantic footnote: It doesn't control the frequency of reads per se, but it should give you the results you are looking for.

chaos39, 2016-07-24:
Hi dtex, thanks for the help !!

closed 2016-07-25 by chaos39