rwaldron/johnny-five #1185: How to use grove Alcohol sensor mq303a in johnny five

chaos39, 2016-07-25:
Hi Guys, I would like to know how to program this kind of sensor in javascript/node js 
Thanks !

dtex, 2016-07-25:
Are you using a Hanwei MQ-303A by itself or a Grove MQ-3?

Link to the product you have would clarify.

chaos39, 2016-07-25:
Hi dtex,
This is the link to the product 
https://www.seeedstudio.com/Grove-Alcohol-Sensor-p-764.html

dtex, 2016-07-25:
Cool. 

This is a bad news/good news situation...

The bad news is that Johnny-Five does yet have support for that specific sensor.

The good news it that you can still do this. All the I2C read and write methods are available on the board.io object so you could roll your own using those helpers, or better yet, add a new sensor type or device to Johnny-Five. Wether it should extend the existing Sensor class or be a new class is something for discussion but it would be a great contribution.

chaos39, 2016-07-26:
may I have link here ? the reason is because i cannot find it. Thanks

chaos39, 2016-07-26:
I currently using sparkfun arduino block to do this sensor

dtex, 2016-07-26:
Doh! Typo.

> The bad news is that Johnny-Five does yet have support

Should read

The bad news is that Johnny-Five does not yet have support

You mean a link to the I2C methods available on board.io? Sure:
https://github.com/firmata/firmata.js#i2c

chaos39, 2016-07-26:
What about using grove mq2? it seems that sensor do sense alcohol too

frxnz, 2016-07-27:
The MQ2 is an analog sensor, so you could use the [`sensor` class](https://github.com/rwaldron/johnny-five/blob/master/docs/sensor.md)

chaos39, 2016-07-27:
I thought mq2 need to warm up for a couple of minutes before using it ? So does that means I can't use this example? https://github.com/rwaldron/johnny-five/blob/master/docs/grove-gas-mq2-edison.md

rwaldron, 2016-07-27:
@chaos39 just wait a couple minutes before using it. 

```js
var five = require("johnny-five");
var Edison = require("edison-io");
var board = new five.Board({
  io: new Edison()
});

board.on("ready", function() {

  // Plug the MQ2 Gas (Combustible Gas/Smoke)
  // module into the Grove Shield's A0 jack
  var gas = new five.Sensor("A0");

  // Plug the Piezo module into the
  // Grove Shield's D6 jack.
  var alarm = new five.Piezo(6);

  gas.scale(0, 100).on("change", function() {
    // Ignore changes for the first 3 minutes
    if (board.millis() < 180000) {
      return;
    }

    if (this.value > 60) {
      if (!alarm.isPlaying) {
        alarm.frequency(five.Piezo.Notes.d5, 5000);
      }
    }
  });
});
```

rwaldron, 2016-07-27:
The product linked above leads to a wiki page that incorrectly describes the wiring for SDA and SCL, implying that it's an I2C device, but then shows code examples with analogRead, which it then elaborates on. If this is just an analog sensor, use the `Sensor` class.

chaos39, 2016-07-28:
Thanks for all the help guys, It really helps me a lot in my project. One more question, is it possible to change the scale of the program gas.scale(0, 100) to gas.scale(100, 2000) ?  I would like  to display the readings on console.log not sure is it possible too

chaos39, 2016-07-28:
if i will be using grove mq2 gas sensor as an alcohol sensor

frxnz, 2016-07-28:
You can scale to whatever values you like. To log the value you can either use `this.value`, or the `change` event is emitted with a value parameter.

```js
gas.scale(100, 2000).on("change", function(value) {

  console.log(this.value); // or console.log(value)

  // ... anything else you want to do

});
```

rwaldron, 2016-07-28:
Just a quick note: I'm hoping to discourage use of `scale(...)` since it changes the entire instance's behavior. It has recently been replaced by `scaleTo`, which simply returns a scaled value from the present value: 

```js
gas.on("change", function() {
  console.log(this.scaleTo(100, 2000));
});
```

closed 2016-07-31 by chaos39