rwaldron/johnny-five #1187: `Board` `exit` event undocument

rhys-vdw, 2016-07-25:
Board's `exit` event is included this [this example](http://johnny-five.io/examples/board-cleanup/), but is not documented [here](http://johnny-five.io/api/board/#events).

I'm not 100% sure of the conditions under which `exit` is fired, nor where that page is grabbing its content from. I will open a PR if someone can help fill me in there.

rwaldron, 2016-07-25:
The "exit" event is undocumented because I was never quite sure about the semantics. It's emitted synchronously on SIGINT. 

If you don't mind doing a docs update, that would be awesome and greatly appreciated. Let me know when it's ready and I will redeploy the site :)

rhys-vdw, 2016-07-27:
@rwaldron took me a while, but I finally worked out where the docs were. [Added a sentence](https://github.com/rwaldron/johnny-five/wiki/Board#events). Hope that's sufficient.

Pretty weird how you can just modify the wiki without any PR/review process...

rwaldron, 2016-07-27:
Ugh, sorry about that :\

> Pretty weird how you can just modify the wiki without any PR/review process...

¯\\\_(ツ)_/¯

I can pull the wiki repo and review locally. I don't know... I didn't make the wiki system. I trust our contributors :)

rwaldron, 2016-07-27:
Nice work!

closed 2016-07-27 by rwaldron

rwaldron, 2016-07-27:
Deployed: http://johnny-five.io/api/board/#events

rwaldron, 2016-07-27:
Thanks again!