rwaldron/johnny-five #1188: Controlling continuous servo motors crashes my app

alexbudin, 2016-07-25:
HI guys,

My setup is this:
- 2 servo motors, cheap ones - Tower Pro - SG-5010
- one Arduino Uno (with StandardFirmata on it)
- one motor driver shield (AdaFruit Motor Shield v1))

Servos are running on pins 9 and 10 of course.
The motor shield is powered separately from 4xAA Batteries (6V)

I have this bit of code here:

```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

   // var servos = new five.Servos([9, 10]);

   // servos[0].to(20);

  var servo1 = new five.Servo({
    pin: 9, 
    type: "continuous"
  });

  var servo2 = new five.Servo({
    pin: 10, 
    type: "continuous"
  });

  // // Clockwise, top speed.
  servo1.cw(1);
  // // Clockwise, top speed.
  servo2.cw(1);

  setTimeout(function(){
    servo1.cw(0);
  },11000)

});
```

The motors start rotating when i run the code, but if i stop the app and run it again, 3 out of 5 times will not run again, and all i get is one of the motors turning for half a second and then none of them moving.

Also when trying to stop any of them:

```
  setTimeout(function(){
    servo1.cw(0);  // or servo1.stop();
  },11000)

```

The app crashes and i get this:

```
1469468151790 Device(s) /dev/cu.usbmodem31
1469468151797 Connected /dev/cu.usbmodem31
1469468155611 Repl Initialized
>> events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: Port is not open
    at SerialPort.write (/Users/alexandrubn/Documents/work_repos/arduino/node_modules/johnny-five/node_modules/serialport/lib/serialport.js:259:24)
    at Board.analogWrite (/Users/alexandrubn/Documents/work_repos/arduino/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:771:18)
    at Board.servoWrite (/Users/alexandrubn/Documents/work_repos/arduino/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:839:20)
    at Servo.Controllers.Standard.servoWrite.value (/Users/alexandrubn/Documents/work_repos/arduino/node_modules/johnny-five/lib/servo.js:73:17)
    at Servo.to (/Users/alexandrubn/Documents/work_repos/arduino/node_modules/johnny-five/lib/servo.js:287:12)
    at Servo.(anonymous function) [as cw] (/Users/alexandrubn/Documents/work_repos/arduino/node_modules/johnny-five/lib/servo.js:492:17)
    at null._onTimeout (/Users/alexandrubn/Documents/work_repos/arduino/routes/paypal_ipn.js:57:12)
    at Timer.listOnTimeout (timers.js:92:15)

```

So i have 2 problems:
1 - the motors sometime start, sometime don't
2 - if i try to stop one the app crashes.

Can someone please help.

Thank you.

Alex

alexbudin, 2016-07-25:
Forgot to add that this is on a mac OSX. The Arduino UNO is connected via USB cable to my mac.

@rwaldron did more testing and it seems it works fine with ONE MOTOR.
```

var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

   // var servos = new five.Servos([9, 10]);

   // servos[0].to(20);

  var servo1 = new five.Servo({
    pin: 9, 
    type: "continuous"
  });

  // var servo2 = new five.Servo({
  //   pin: 10, 
  //   type: "continuous"
  // });

  // // Clockwise, top speed.
  servo1.cw(1);
  // // Clockwise, top speed.
  //servo2.cw(1);

  setTimeout(function(){
    servo1.stop();
  },11000)

});
```

And it seems to be working every time i run it.

This is getting so confusing :( I hope someone can help me.

dtex, 2016-07-25:
The servo pins are not powered off the 4 AA batteries, they are using the same 5V supply that drives the Uno (1). At some point the second servo is pushing the total current draw up over 500mA causing the Uno to brown out and be unreachable. 

That motor shield power is just for the DC motors or stepper. If you're not going to be using a DC or stepper motor, you don't need the shield at all.

(1) https://learn.adafruit.com/adafruit-motor-shield/power-requirements#how-to-set-up-the-arduino-plus-shield-for-powering-motors.

alexbudin, 2016-07-25:
@dtex I was just reading that now :) So annoying that it doesn't power the servos also.

I was noticing the LED on the driver board dimming a bit after the second one would start.

Thank you Donovan, really helpful.

I will just power them separately and get rid of the board.

Alex

closed 2016-07-25 by dtex