rwaldron/johnny-five #1192: What is the whole environment for running johnny-five?

iszufu, 2016-07-30:


rwaldron, 2016-08-02:
The serialport bindings in node-serialport (which are used by Johnny-Five) are compiled at npm-install-time, or installed using a remote pre-compiled binary. If the binary does match the platform/architecture, then the code isn't going to work correctly. The correct way to share apps like this, is to allow them to correctly install their dependencies for the target platform. 

@noopkat do you have any advice for packaging nw apps?

noopkat, 2016-08-02:
If you're transferring the app as just the code source from Github, then manually running the rebuild steps on each computer will be necessary, but it doesn't sound like you're trying to do that.

If you're moving a fully packaged app from computer to computer (eg. an nwjs .exe file, or a .app file, etc) and the computer has the exact same architecture and operating system major version, then I would assume that the app should work perfectly as it did on the original machine. I have successfully packaged apps for Windows, Mac and Linux for distribution in this manner and they have worked, as long as you have packaged the Windows version on Windows, etc. It works because the packaged app has everything precompiled and ready to go. 

With serialport specifically, having a matching operating system + architecture should be all thats required in order to have the pre-packaged app work out of the box. This is from the anecdotal experience of releasing one app for OSX, one app for Windows, and one app for Linux with everything working fine on a variety of machines running those operating systems. There might be a difference when it comes to `noble`, or whichever nodejs bluetooth dependency is being used for your app.

My experience is with USB communication, and not Bluetooth. Are perhaps there nuanced differences in machines having the right bluetooth dependencies outside of the nwjs app? Beyond drivers, perhaps. If there's a way for the packaged app to log errors or report them visibly, that might help with why.

Another way to try to debug is I'm wondering if installing the source code / repo on a different machine and then manually recompiling native modules + the app package itself would bring about any useful errors at runtime that you could work back from?

rwaldron, 2016-08-23:
Is this sufficiently answered? Can we close? Let me know :)

iszufu, 2016-09-08:
Thank you for your answer. Now I can already successfully run my nwjs-based application on win10 system. However, when moving the app to win7 system, some parts of computers can run it correctly, but some can not. For example, my arduino car would only move one side of wheel when I hope it going forward, which happens randomly, rather than in 100 percents. I think the problem might be caused by the high version of node runtime, therefore I am trying to reduce it to be 0.10.22 (corresponding to [nwjs version of v0.8.6](https://github.com/nwjs/nw.js/blob/nw17/CHANGELOG.md)), hope it will fix the probelm. Can you give me some advices to fix it?
Here is another question. Cause in my application, only some funtions that related to motor controlling being used, but many other modules are not. So, is it possible to cut the useless modules of johnny-five to decrease the size of whole application.

closed 2018-08-28 by stale[bot]