rwaldron/johnny-five #1196: No data when reading light from the TSL2561 (RPI 2g, Arduino Uno)

larskaare, 2016-08-05:
Greetings,

I am experimenting with an Adafruit TSL2561 sensor but are not able to get any light values.

I have the current setup; RPI with J5, and Arduino Uno connected using USB, running firmata on the Arduino, having the TSL2561 sensor wired with default settings all the way. The sensor i using i2c - default addresses. My experiment is based on the default example for the tsl2561 sensor.

Testing that I have done:
* I have tested the Arduino & the wiring accessing the sensor from the Arduino ide. This works fines. I am getting good readings.
* I have tested the RPI->J5->Arduino->Firmata link by hooking some simle leds. This works fine, J5 talks to the Arduino and controls the leds.
* I have followed the J5 code from the light.js-> into firmata.js and can see that the tsl2561 code for initializing is running and that i2c calls are made from firmata for setting up the sensor. This looks ok; the sensor respons on the default address as expected, timing and gain messages are sendt into the i2c bus and the code for read executed.  This where things are stopping now.

In the light.js code i have tried setting the frequency of the sensor and using the on("data") event. The data that is returned is always [0,0,0,0]. The on("change") event never triggers.

Putting it briefly - I am not getting values back from the sensor - as far as I can see that is :)

I have done some debugging, but it getting quite complex with callbacks on callbacks, events and emitters, i2C over firmata - so any guidance would be appreciated.

larskaare, 2016-08-07:
Did some more testing, added a BME280 sensor to the setup and configured it to communicate over i2c. It works like a charm.

rwaldron, 2016-08-23:
> Did some more testing, added a BME280 sensor to the setup and configured it to communicate over i2c. It works like a charm.

Fascinating... I wonder what the root cause was?? Can we close?

larskaare, 2016-08-24:
Hi - regretfully i have not succeeded in reeding any values from the tsl2561 over J5-Arduino yet. I have confirmed that the wiring are correct and working, and that the sensor is ok - by by making my own program in Arduino. I have also confirmed that J5 and the firmata/I2C seems to work by using J5 to read over firmata/I2C on the Arduino for the temp sensor (BME280). 

I have not succeeded with J5 -> Firmata and the tsl2561 and would appreciate some guidance on how to debug this issue, or things to try out :)  (In the mean time I am getting light values from a standard photoresistor - so this is not the most important thing in the world)

rwaldron, 2016-08-24:
@larskaare ok, I just wired up a tsl2561 and your issue is 100% reproducible

rwaldron, 2016-08-24:
@larskaare in your project, do this: 

```
npm install rwaldron/johnny-five#1196
```

The fix addresses the missing timing enforcement. I must've had this when I was developing and then removed it thinking I was simplifying or improving? I have no idea. I tested this by shining a flash light back and forth over the sensor and got really nice responses. Let me know how it goes

larskaare, 2016-08-27:
Thanx! No success yet :) This is what i did; I wired up the sensor to the arduino, loaded up a program to the arduino that set-up the sensor and printed the lux values (an example from adafruit) - this verified the sensor and the wiring. I then loaded the standard firmata up to the arduino. After this I created a small project that installed the j5 module as instructed (I verified that your fixes was in the light.js file) and wrote a small rutine the defined the light sensor with the proper controller and tried to read the values from the sensor on the value and level properties. My code was based on the examples in the j5 api documentation. It always return 0 on the "data" event ..... and never triggers the "change" event .....

rwaldron, 2016-08-29:
> My code was based on the examples in the j5 api documentation. It always return 0 on the "data" event ..... and never triggers the "change" event .....

Can you describe how you attempted to affect the environment to trigger changes in the value read? ie. did you flash a light over the sensor, or something similar?

larskaare, 2016-08-29:
@rwaldron, amongst others I used the "flash light" on my iPhone. 

When using the specific code running on the Arduino the sensor was responding ok to he ambient light. I experimented with covering up the sensor, turning off the light in the room - all with the sensor responding as expected. (As a side note, after my last post i did a quick test to check that the sensor was on the expected address on I2C used by J5 - which it was).

rwaldron, 2016-08-30:
I wonder if J5 is using some configuration that limits its sensitivity? Can you share the Arduino code that you're using so that I can compare? Thanks!

rwaldron, 2016-08-30:
(Also, I really appreciate your patience while we work through this together)

larskaare, 2016-08-31:
I used two Arduino libraries from Adafruit; the unified sensor driver (https://github.com/adafruit/Adafruit_Sensor) and then the specific driver for the TSL2561 (https://github.com/adafruit/Adafruit_TSL2561). The example in the specific driver is what ended up on my Arduino - which I referred to earlier.

rwaldron, 2016-08-31:
@larskaare perfect, this is very useful—I'm going to spend some time on this right now and will report back when I've learned something.

rwaldron, 2016-08-31:
Check this out...

- https://github.com/adafruit/Adafruit_TSL2561
- https://github.com/adafruit/TSL2561-Arduino-Library

0_0

rwaldron, 2016-09-06:
I'm not going to have a chance to write the full tests that I want to write until probably next week, but I want you to test the code anyway, so I'm going to push the branch for you to work with

rwaldron, 2016-09-06:
```
npm install rwaldron/johnny-five#1196
```

larskaare, 2016-09-07:
I've done some preliminary testing. I am getting lux values from the sensors. I intend to do some more testing comparing those values to the ones I'm getting while using the "native" Arduino routine - but this will most likely take some time :) I've also been testing a setup with this TSL2561 and a BME280 combined. Both are on the i2c bus and seems to work fine together.

larskaare, 2016-09-28:
@rwaldron just a short status. I'm planning to do some more testing here, but my TSL2561 got hijacked  by another project so I had to order a few new ones :)

rwaldron, 2016-10-04:
@larskaare any updates?

larskaare, 2016-10-05:
@rwaldron not yet - i've set aside some time for tomorrow and intend to do a bit more thorough testing then. I guess i'm still testing the 1196 branch....

rwaldron, 2016-10-05:
Yes please! I'm waiting for you to sign off on the fixes :)

larskaare, 2016-10-06:
Done more testing on #1196 using the following protocol (all tests done with default setup, no options used)
* Testing on events "change" and "data"
* Environment; ambient, iPhone flash and dark (sensor partly covered with hand)
* Testing on this.value and this.lux

Tests

1.  Using TSL2561 alone. This.value (0.01, 0.03, 0.00) (ambient, flash, dark), lux (335, 500-0, 112). Sensor is working ok with J5, responding to changes. We should think about how to scale this.value and it also seems like the J5 is giving 0 when the light is to strong?

1. Using TSL2561 with another sensor on the I2C bus (BME280). Gives similar results as for test 1. The BME280 sensors are working ok giving good output on temperature, humidity and pressure

1. Adding a analog sensor to test 1 and 2. Comparable results. Test ok

1. Testing TSL2561 on the Arduino directly (using code from Adafruit) to compare values. I get the following readings on lux (332-333, 500-1400, 17-20). The values are comparable except for iPhone flash where J5 gives "0" on strong light while the Adafruit code goes up to approx 1400. 

1. Testing the branch and TSL561 on my hobby project (happyeye/happypi - https://github.com/Statoil/happyeye) where a lot of nasty stuff is taking place. I get good results from the TSL2561 and the nasty stuff still seems to be nasty. Test passed.

In essence
* this.value scaling (minor issue)
* this.lux does not handle strong light (gives zero)

We should investigate the "strong light" giving zero "issue" @rwaldron

rwaldron, 2016-10-06:
This is awesome feedback! I will iterate, with specific focus on the final lux calculation and ping you when ready for more review.

closed 2016-11-01 by rwaldron