rwaldron/johnny-five #1197: Johnny Five not connecting to Arduino from Raspberry Pi

Oddboxx, 2016-08-06:
Good morning/afternoon/evening/day!

I have started a new project using Johnny five for the first time and I've hit a snag.

My plan is to run node and Johnny Five from my Raspberry Pi (2B) to control/interact with a USB connected arduino.

I have run the hello world code from my desktop linux machine which ran fine with the StandardFirmatePlus recommended in your documentation.

However, the same code running off of the raspberry pi is not as successful.

The board is never connected to; see my copy-paste:

```
pi@raspberrypi  $ node index.js 
1470510380725 Device(s) /dev/ttyACM0  
pi@raspberrypi  $ 
```

As you can see, it never connects to the board and drops out before reaching the REPL.

When I pass the port into the setup as a parameter:

`new five.Board({port:"/dev/ttyACM0"});`

I get a JS error: 

```
module.js:339
    throw err;
    ^

Error: Cannot find module 'firmata'
    at Function.Module._resolveFilename (module.js:337:15)
    at Function.Module._load (module.js:287:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Board.Serial.connect (/home/pi/sprout-firmware/node_modules/johnny-five/lib/board.js:105:14)
    at new Board (/home/pi/sprout-firmware/node_modules/johnny-five/lib/board.js:287:22)
    at Object.<anonymous> (/home/pi/sprout-firmware/index.js:14:13)
    at Module._compile (module.js:435:26)
    at Object.Module._extensions..js (module.js:442:10)
    at Module.load (module.js:356:32)

```

my node modules are Johnny-Five and Serialport. 

Any help or even a response would be appreciated!

monteslu, 2016-08-06:
Try running with sudo.  Might not have permission on the serial port.

Oddboxx, 2016-08-06:
> running with sudo

Same behaviour I'm afraid to say :(

rwaldron, 2016-08-07:
> Error: Cannot find module 'firmata'

Did you `npm install johnny-five` _on_ the RPi, or move it from somewhere else?

Oddboxx, 2016-08-07:
I cloned it onto the Pi from my desktop.

I installed johnny five again directly on the Pi in a new project and it works! What a simple solution! Sorry for raising an issue for such a simple problem. I'm not much of a wiz with node.

Hopefully somebody else might learn from my mistake.

closed 2016-08-07 by Oddboxx