rwaldron/johnny-five #1198: Using Arduino's TX/RX via firmata.js

Mrparkers, 2016-08-08:
Hi all,

I'm looking to use my Arduino's TX and RX pins to communicate to a Raspberry Pi using a radio transceiver that I bought.  I was looking around the docs for firmata.js and I found [this](https://github.com/firmata/firmata.js/#serial) which seems to imply that I should be able to do this.

After looking through the j5 docs, it doesn't look like this is officially supported yet (please correct me if I'm wrong here).  So I guess my question is - Is there a good way to gain access to the underlying firmata instance that j5 uses so I can have access to `board.serialWrite` or its friends?

Alternatively, is there a better way to approach something like this?

Thanks!

dtex, 2016-08-08:
Those methods are available via the ```board.io``` object so ```board.io.serialWrite()```

> After looking through the j5 docs, it doesn't look like this is officially supported yet

The serial* methods are part of firmata not johnny-five, so they wouldn't be documented in the J5 repo.

> Alternatively, is there a better way to approach something like this?

Two questions:

You're running node.js on the Pi and you want to talk to an Arduino, Is that correct? 

Which transceiver are you using?

Mrparkers, 2016-08-08:
Yeah, I'm running Node on my Pi and I would like my Pi to talk to my Arduino.

[Here](https://www.amazon.com/gp/product/B013UVNVIS) is the transceiver I bought.  I've used this same transceiver before to allow a laptop to communicate to a Pi, so I'm pretty sure it should work for this too.

dtex, 2016-08-08:
I'll admit to not knowing much/anything about this transceiver but I would think it just lives inline and replaces the hardwired serial connection. I wouldn't expect that you have to do anything other than explicitly [specify a port](http://johnny-five.io/api/board/#parameters) when initializing your board, assuming it has configured with another app/utility.

Mrparkers, 2016-08-08:
> it just lives inline and replaces the hardwired serial connection

I'm not sure if I'm understanding what you're saying here.  I might not have explained my problem all that well.

My goal isn't to replace the hardwired serial connection that the Arduino uses to receive instructions from the host, but to use this transceiver alongside the hardwired connection.

In other words, right now I have Laptop -> Blue USB Cable -> Arduino, which I can use to move servos around, blink LEDs, etc.  What I want to achieve is Laptop -> Blue USB Cable -> Arduino -> Transceiver 1 -> Transceiver 2 -> Raspberry Pi.  I was hoping that I could connect the first transceiver to the Arduino's TX/RX pins to talk to the Pi without removing the USB cable that is responsible for sending the interpreted instructions to the Arduino via firmata.

Does this make sense?  And would I still be able to use `board.io.serialWrite` to do this?

dtex, 2016-08-08:
Oh yes, that's much more clear. 

**Take everything below this line with a grain of salt. I've never done this.**

I think it's possible to use board.io.serialWrite to do this. If this is an Uno/Uno clone you will only have one hardware serial port (the one connected to your laptop) so you will have to use software serial which firmata handles just fine. As long as the baud rate can be dialed down to < 57,600 it shouldn't be an issue and if you are using an Arduino with more hardware serial ports (i.e. a Mega) you can use hardware serial which is faster.

I believe you will need a serial <-> USB adapter between the Arduino and the transceiver.

For the Pi side you would use node-serialport to listen/send to the Arduino. Johnny-Five & firmata would just live on the laptop/Arduino.

dtex, 2016-08-08:
... and I do not think that this will be easy but I do think it's possible.

Mrparkers, 2016-08-11:
Thanks for the info!  Unfortunately, one of my transceivers appears to have arrived DOA (TX pin doesn't work), but I'll be sure to update here when I get a new one and try this out, if you're interested in how I get it working.

rwaldron, 2016-08-23:
> Transceiver 2 -> Raspberry Pi

Can you link to the transceiver you're using?

mMerlin, 2016-08-24:
@rwaldron The link in the third comment of this thread seems to be what you want
https://www.amazon.com/gp/product/B013UVNVIS

rwaldron, 2016-08-24:
@mMerlin good eye, I totally missed that.

shmrymbd, 2017-10-02:
hi anyone can share any sample code?

soundanalogous, 2017-10-02:
See the hw-serial-read-gps.js and sw-serial-read-gps.js examples here: https://github.com/firmata/firmata.js/tree/master/examples.

closed 2019-02-04 by stale[bot]