rwaldron/johnny-five #1199: Connecting multiple boards to J5 - Issues

criticalboot, 2016-08-12:
We are attempting to connect 2 boards to J5 via ethernet over 2 ports (3015 & 3016). 

**Scenario 1:** 
When both the boards are connected, the board on port 3016 ( 2nd board) is not responding. But board on 3015 (1st board) is working as expected. We are not sure what is happening internally.

**Scenario 2:**
When we have board 1 powered Off and have board 2 On, the board is not getting instantiated, and throwing the following error 

`Uncaught TypeError: Cannot read property 'type' of null`

We are trying to understand the following as to what is happening in both the cases above.

**What do we actually want?**
a. We need multiple boards to be able to connect to J5 independently with out intervention from other boards.
b. We are using the 'boards' object, where if one of the boards errors out leading to application crashing. This is a very practical problem for our solution design where when the application crashes our whole solution will come down leading to massive losses. 

Please suggest a solution out for this.

rwaldron, 2016-08-23:
- Can you provide more details about the hardware itself? What boards? What ethernet hardware?
- There is no built-in ethernet support in Johnny-Five, then I assume you're using [Etherport](https://github.com/rwaldron/etherport) or something similar?

criticalboot, 2016-08-24:
- We using Atmel 644PA based Board, with ESP8266.
- Yes,  we are using Etherport.

rwaldron, 2016-08-24:
I've never used either of those (and of course, don't have them on hand). The policy of this project has always been that I don't claim something is supported unless I've personally verified it, but I'm always trying to verify as many boards, as often as possible. 

- w/r to Atmel 644PA, what is the exact board you're using, so I can try to acquire two of them
- same for ENC28j60

Can you link me to a site where I can purchase these? Thanks

criticalboot, 2016-08-25:
Thank you for your response
Please find the boards that we are using below.

1. Micro Controller : 644 PA - AU  - https://www.microduino.cc/store  Look for  **"Microduino core+ (ATmega644PA@16M,5V) "**

2. [ENC28J60](http://www.ebay.com/itm/1PCS-MiNi-ENC28J60-Ethernet-LAN-Network-Module-For-Arduino-SPI-AVR-PIC-LPC-/381447240602?hash=item58d006fb9a:g:A5YAAOSwT5tWKuQ~)

rwaldron, 2016-08-25:
Ok, I ordered two of each. Looks like the Microduino Core+ is going to take quite some time to get here—sorry :\ I appreciate your patience in the meantime.

criticalboot, 2016-08-26:
No problem. We will wait for you to get back with your testing results.

closed 2018-08-28 by stale[bot]