rwaldron/johnny-five #1200: Documentation for i2cConfig is incorrect/confusing

nkolban, 2016-08-14:
If we look at the documentation for i2cConfig() found here:

http://johnny-five.io/api/board/

We find that its documented signature is:

`i2cConfig([milliseconds])`

while when we look here:

https://github.com/rwaldron/io-plugins

We see that the signature of i2cConfig() is:

`i2cConfig(options)`

with a rich set of options available.  This issue asks for examination and consolidation of docs.

rwaldron, 2016-08-15:
Thanks for the report!

Since I wrote both things that you're referring to, I can assure you that neither is wrong :P 

> This issue asks for examination and consolidation of docs

IO-Plugins and Johnny-Five docs are not one in the same. `i2cConfig(...)` _can_ accept just a number, or an object of options. This is simply a case of having too much work and not enough time or personal bandwidth to do everything. 

> but I was unable to find any git source for the public web site pages.

[It's all in the wiki for this repo](https://github.com/rwaldron/johnny-five/wiki/Board). 

I've just updated the Board docs to include the `i2cConfig(options)` signature, with a link to the IO-Plugins page as well.

closed 2016-08-15 by rwaldron

nkolban, 2016-08-16:
Is the web page fed from the Wiki content?  If potential contributors like myself update the Wiki, will that make it to the web page?

Am falling in love with Johnny-five :-)

rwaldron, 2016-08-16:
> Is the web page fed from the Wiki content? 

It sure is :)

> If potential contributors like myself update the Wiki, will that make it to the web page?

Yes sir!

> Am falling in love with Johnny-five :-)

I'm so happy to read this! :D !