rwaldron/johnny-five #1202: Feature: Boolean argument for `Led.toggle`

lyzadanger, 2016-08-19:
Hi! I have always found the optional `Boolean` parameter for jQuery's [toggle](http://api.jquery.com/toggle/#toggle-display) method quite handy and occasionally wish it existed on `Led` (and `Led.RGB`, etc.). That is, if a Boolean argument is present, `toggle` sets the component to that state. This can be useful for simplifying logic.

I'd be happy to implement, test and PR if this is something that would be a good idea.

Resseguie, 2016-08-19:
I'm not opposed to that. I remember in the past thinking that option would have simplified my logic as well. (I'd have to go digging to find concrete examples.)

rwaldron, 2016-08-23:
> This can be useful for simplifying logic.

Can you provide an example?

lyzadanger, 2016-08-23:
@rwaldron Sure, something like this:

```javascript
// ...
var photoresistor = new five.Light(/* ... */);
var led = new five.Led(/* ... */);
photoresistor.on('change', () => {
  led.toggle((photoresistor.level > 0.5));
});
```

rwaldron, 2016-08-23:
I see, and I disagree that `toggle(...)` should learn new tricks: `toggle` is `toggle`. What you're looking for is what `led.update(...)` already does, except that method doesn't know anything about   boolean values—but it does do operations that will result in its inputs being coerced to numbers. Please try that and let me know.

lyzadanger, 2016-08-23:
@rwaldron 

> Please try that and let me know.

Do you mean "try it as it is now" or "make changes to `update` so it behaves in the way you're looking for?"

I agree that on face value, adding a Boolean argument to `toggle` makes it semantically inaccurate, but was following jQuery-style precedent. That's where that came from. In any case, yeah, I'd love to find some method on `Led` that can play this role...it'd be nice!

rwaldron, 2016-08-23:
> "try it as it is now"

👍

lyzadanger, 2016-08-24:
Welp, I'll close this for now, then.

closed 2016-08-24 by lyzadanger