rwaldron/johnny-five #1203: EventEmitter.emit('error') is reserved.

overflowz, 2016-08-20:
Hi, I faced a rough problem here. I was trying to construct simple Thermometer class (DS18B20) and forgot to actually plugin sensor itself, so I get error:

`Error: FAILED TO FIND TEMPERATURE DEVICE`

But, somehow my application was crashing, saying:
````
events.js:160
      throw er; // Unhandled 'error' event
````

so I was looking up in a source and found this piece:
````
EventEmitter.prototype.emit = function emit(type) {
  var er, handler, len, args, i, events, domain;
  var needDomainExit = false;
  var doError = (type === 'error');

 events = this._events;
  if (events)
    doError = (doError && events.error == null);
  else if (!doError)
    return false;

  domain = this.domain;

  // If there is no 'error' event listener then throw.
  if (doError) {
    er = arguments[1];
    if (domain) {
      if (!er)
        er = new Error('Uncaught, unspecified "error" event');
      er.domainEmitter = this;
      er.domain = domain;
      er.domainThrown = false;
      domain.emit('error', er);
    } else if (er instanceof Error) {
      throw er; // Unhandled 'error' event
    } else {
      // At least give some kind of context to the user
      var err = new Error('Uncaught, unspecified "error" event. (' + er + ')');
      err.context = er;
      throw err;
    }
    return false;
  }
...
````

As you can see, it checks if type is 'Error' then checks if it has domain applied, if not, then it's uncaught exception and should be handled globally with process.on() function and I don't quite like that.

Any thoughts?

rwaldron, 2016-08-23:
Thanks for filing!

> I was trying to construct simple Thermometer class (DS18B20) and forgot to actually plugin sensor itself, so I get error:

> `Error: FAILED TO FIND TEMPERATURE DEVICE`

> But, somehow my application was crashing, saying:
> ````
> events.js:160
>       throw er; // Unhandled 'error' event
> ````

I read this as: our use of error events worked as intended ;)

closed 2016-08-23 by rwaldron

overflowz, 2016-08-23:
Actually not. "error" event is reserved, it breaks whole event loop instead of throwing error.

I also found this article over google: http://blog.yld.io/2015/12/15/using-an-event-emitter/#.V7zVEZh96Uk

quote from there:
````
As we've seen, an event type is defined by an arbitrary string. In that aspect, all events are treated equally, with one exception: errors.

If an event emitter emits one event for which it has no listeners, that event gets ignored. But, in the special case of the event name being "error", the error is instead thrown into the event loop, generating an uncaught exception.
````

rwaldron, 2016-08-23:
No, there's nothing reserved about it, it just has additional semantics. If you read the docs https://nodejs.org/api/events.html#events_error_events it's very clear. The behavior in J5 is correct and as intended. The program should fail if the sensor is not connected and there is no error handler

overflowz, 2016-08-24:
Oh, my apologies then! I wanted something like:
````
var thermo = new five.Thermometer({ ... });
thermo.on('error', function () { ... }); // never fires up
````

Thanks for the info anyways!

rwaldron, 2016-08-24:
Yep, that's correct