rwaldron/johnny-five #1204: Mismatched Module Version when attempting to build against Electron

Cryptoc1, 2016-08-20:
After rebuilding electron using `./node_modules/.bin/electron-rebuild` I get the error: `Module version mismatch: Expected 49, got 48`. I tried running `./node_modules/.bin/electron-rebuild -n 48`, as well as `./node_modules/.bin/electron-rebuild -n 49`, but neither worked.

I've attempted using the `rm -rf node_modules/` method, then running `npm install; ./node_modules/.bin/electron-rebuild`, and I've even tried force clearing the npm cache with no luck.

At the moment I'm using an Arduino Uno, but will soon be switching to a Micro. As far as usage of Johnny-five goes, I'm only trying to initialize a board when electron loads (within the `app.on('ready', function...)` for electron's API). My electron app is a tray/status bar app, so johnny-five is being used in the main process of electron, not the render.

```
node: v6.3.1
electron: v1.3.3
npm: v3.10.3
johnny-five: ^0.10.0
```

nlemoine, 2016-08-23:
I had a similar issue, seems like Johnny Five isn't compatible with electron-prebuilt > 1.0

Cryptoc1, 2016-08-23:
I bit of a temporary workaround I'm using at the moment is spawning a child process to run a node script that implements my needs of Johnny-five. I'm thinking of moving to a more "client-server" model though. This may not work for your specific needs, but it's good enough for me. 

Hopefully this issue will be worked on soon.

rwaldron, 2016-08-23:
Thanks for filing an issue, and I'm happy to try to help get this sorted, but I'm not sure what any of this actually has anything to do with Johnny-Five. 

Can you both follow this: http://meow.noopkat.com/using-node-serialport-in-an-electron-app/ and report back? Thanks!

nlemoine, 2016-08-27:
Thanks for your feedback and help. 
I already stumbled upon this article with no success. 

Running Node 6.2.1 with NVM on Mac OS X 10.11.6, I executed the following commands.

Install the quick start Electron starter project:
```
npm install -g electron-prebuilt
git clone https://github.com/electron/electron-quick-start
cd electron-quick-start
npm install
```

I followed the article steps: 
```
npm install --save serialport
npm install --save-dev electron-rebuild
./node_modules/.bin/electron-rebuild -$(electron -v)
```

I pasted the given code in the `index.html` file:
```
<script>  
  var sp = require('serialport');

  sp.list(function(err, ports) {
    console.log(ports);
  });
</script>
```

And started the app with ```npm start``` and still have the same issue:

[image]

drj-io, 2016-09-07:
I was getting the same error in my electron app, that FORKS another node app and runs Johnny Five (for other reasons listed in #1133) 

The example above worked for me as follows:

```
cd node_modules/serialport 
sudo HOME=~/.electron-gyp node-gyp rebuild --target=1.3.4 --arch=x64 --dist http://atom.io/download/atom-shell
```
this resolved the mismatch issue for me quickly.

nlemoine, 2016-09-07:
It actually worked! Thanks a lot @drj-io!
👍

nickstew, 2016-09-28:
Thanks a ton @drj-io!  I did a write up on my steps to fix this issue in [opensprints-electron issue #5](https://github.com/opensprints/opensprints-electron/issues/5).  Hope my issue writeup helps find a permanent solution to this problem.

closed 2018-08-28 by stale[bot]