rwaldron/johnny-five #1205: How to use DHT11Feature?

overflowz, 2016-08-20:
Hi, as far as you know, I'm new on such things. I was wondering how can I use the DHT11Feature with ConfigurableFirmata? I just merged DHT11Feature with ConfigurableFirmata but now, I don't know how should I get the data from it.

I've tried this so far, but of course, callback does not trigger at all cause I'm not really sure what should I do. Could anyone point me out? Thanks!
````
var SerialPort = require("serialport");
var five = require("johnny-five");
var board = new five.Board({
    port: new SerialPort("COM6", {
        baudrate: 57600,
        buffersize: 1
    }),
    repl: false
}).on('error', function () {
    console.log('error');
}).on('ready', function () {
    this.io.sysexResponse(0x7E, function (data) {
        console.log('got data');
    });
});
````

overflowz, 2016-08-21:
never mind, had to use
``
 this.io.sysexResponse(0x00, function (data) {
        console.log('got data');
    });
``

closed 2016-08-21 by overflowz

rwaldron, 2016-08-23:
I'm curious about this: 

```js
var SerialPort = require("serialport");
var five = require("johnny-five");
var board = new five.Board({
    port: new SerialPort("COM6", {
        baudrate: 57600,
        buffersize: 1
    }),
    repl: false
})
```

Why are you explicitly using Serialport? If you're connecting a board directly to your pc, then this should work fine: 

```js
var five = require("johnny-five");
var board = new five.Board({ port: "COM6" });
```

overflowz, 2016-08-23:
Yes, this thread makes no sense, it was just my mistake. You can delete if it's possible though.

Regards

rwaldron, 2016-08-23:
But everything worked out alright?