rwaldron/johnny-five #1206: Device or Firmware Error [Pro Mini] [windows 7]

yurypaleev, 2016-08-21:
```
1471768315851 Connected COM3  
events.js:146
1471768325858      throw err;
       ^
Device or Firmware Error
 Error: Uncaught, unspecified "error" event. ([object Object])
A timeout occurred while connecting to the Board. 
    at Board.emit (events.js:144:17)

    at Board.log (C:\Users\Admin\WebstormProjects\esp-j5\node_modules\johnny-five\lib\board.js:625:8)
Please check that you've properly flashed the board with the correct firmware.
    at Board.(anonymous function) [as error] (C:\Users\Admin\WebstormProjects\esp-j5\node_modules\johnny-five\lib\board.js:636:14)
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

    at Board.<anonymous> (C:\Users\Admin\WebstormProjects\esp-j5\node_modules\johnny-five\lib\board.js:385:14)
If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.     at Timer.listOnTimeout (timers.js:92:15)
```

rwaldron, 2016-08-23:
Does the Pro Mini have StandardFirmata uploaded to it?

yurypaleev, 2016-08-23:
Yes

rwaldron, 2016-08-23:
Ok, thanks, I'll have to try and repro

rwaldron, 2016-08-24:
I'm unable to reproduce this and it just occurred to me that you should verify that COM3 is actually the correct port. Can you check in the hardware manager, find the USB hardware and make sure that COM3 is actually the port in use? Thanks

yurypaleev, 2016-08-25:
I uploaded the StandardFirmata using Arduino IDE using COM3 port

reconbot, 2016-08-26:
(note: the comport can change)

You'll want to check power and the serial console cable connections. Did you try to restart the device during connection attempts? Being able to flash the arduino pro mini (is it official or a clone?)  is a good sign.

yurypaleev, 2016-08-26:
@reconbot: I have the automatic uploader. It the Arduino clone.

reconbot, 2016-08-27:
Can you link me to it?

yurypaleev, 2016-08-28:
@reconbot: I do not understand..

reconbot, 2016-08-28:
Do you have a link to the product page for your arduio clone? I'm betting it's not automatically restarting when the connection is made. Most arduinos do that. I'm hoping there's information about it on the product page.

If you haven't you should try restarting your arduio when it tries to connect.

yurypaleev, 2016-08-28:
No, I do not have..
But why, then, I do not shake restart when a upload StandartFirmata?

sant123, 2016-11-28:
@yurypaleev please check the correct port of your Arduino using de Arduino IDE. In tools - > ports. I have my Arduino set on port 7. That's why I did this in my code:

``` javascript 
var board = new five.Board("COM7");
```

Cheers 🍻

closed 2018-08-28 by stale[bot]