rwaldron/johnny-five #1209: switching off the program and switching on the program

chaos39, 2016-08-25:
Hello guys, I currently doing my school project, I am trying to use intel Edison together with Arduino block to fetch readings and send those readings to the server and the server can switch on and off my program by using a mobile phone. But I am not sure how to do it. I was wondering if there is anyone that knows how to do it.

Thanks!
This is my code that I Currently doing  

var amqp = require('amqplib/callback_api');
var five = require("johnny-five");
var cmdflag = 0;
var Edison = require("edison-io");
var board = new five.Board({
  port: "/dev/ttyMFD1"
});

var CLOUDAMQP_URL = 'amqp://james:1234@192.168.0.99:5672/%2F';

var amqpConn = null;

function start() {
  amqp.connect(CLOUDAMQP_URL + "?heartbeat=60", function(err, conn) {
    if (err) {
      console.error("[AMQP]", err.message);
      return setTimeout(start, 1000);
    }
    conn.on("error", function(err) {
      if (err.message !== "Connection closing") {
        console.error("[AMQP] conn error", err.message);
      }
    });
    conn.on("close", function() {
      console.error("[AMQP] reconnecting");
      return setTimeout(start, 1000);
    });
    console.log("[AMQP] connected");
    amqpConn = conn;
    whenConnected();
  });
}

function whenConnected() {
  startPublisher();
  startWorker();
}

var pubChannel = null;
var offlinePubQueue = [];
function startPublisher() {
  amqpConn.createConfirmChannel(function(err, ch) {
    if (closeOnErr(err)) return;
      ch.on("error", function(err) {
      console.error("[AMQP] channel error", err.message);
    });
    ch.on("close", function() {
      console.log("[AMQP] channel closed");
    });

    pubChannel = ch;
    while (true) {
      var m = offlinePubQueue.shift();
      if (!m) break;
      publish(m[0], m[1], m[2]);
    }
  });
}

function publish(exchange, routingKey, content) {
  try {    
    pubChannel.publish(exchange, routingKey, content, { persistent: true },
                      function(err, ok) {
                        if (err) {
                          console.error("[AMQP] publish", err);
                          offlinePubQueue.push([exchange, routingKey, content]);
                          pubChannel.connection.close();
                        }
                      });
  } catch (e) {                                                                                                                               
    console.error("[AMQP] publish", e.message);
    offlinePubQueue.push([exchange, routingKey, content]);
  }
}
// A worker that acks messages only if processed succesfully
function startWorker() {
  amqpConn.createChannel(function(err, ch) {
    if (closeOnErr(err)) return;
    ch.on("error", function(err) {
      console.error("[AMQP] channel error", err.message);
    });

    ch.on("close", function() {
      console.log("[AMQP] channel closed");
    });
    
    ch.prefetch(10);
    ch.assertQueue("CupIN", { durable: true }, function(err, _ok) {
      if (closeOnErr(err)) return;
      ch.consume("CupIN", processMsg, { noAck: false });
      console.log("Worker is started");
    });
    
    function processMsg(msg) {
      work(msg, function(ok) {
        try {
          if (ok)
            ch.ack(msg);
          else
            ch.reject(msg, true);
        } catch (e) {
          closeOnErr(e);
        }
      });
    }  
  });
}
    function work(msg, cb) {
    console.log("Received:", msg.content.toString());
    var msgIn = msg.content.toString();
    colonIndex = msgIn.indexOf(":");
    if (colonIndex>0){
        cmdString = msgIn.substr(colonIndex+2, 3);
        switch(cmdString){
            case"OFF":
            console.log("Command:",cmdString);
            cmdflag = 0;
            break;
            case"ON":
            console.log("Command ：",cmdString);
            cmdflag = 1;
            break;
            default:
            console.log("Received:",cmdString);
}
  
    }
  cb(true);
}

function closeOnErr(err) {
  if (!err) return false;
  console.error("[AMQP] error", err);
  amqpConn.close();
  return true;
}

board.on("ready", function() {
 
var lcd = new five.LCD({
    controller: "JHD1313M1"
  });
  var led0 = new five.Led(10); // yellow led
  var led = new five.Led(11);
  var led1 = new five.Led(12);
var gas = new five.Sensor({
    controller: "GROVE",
    pin: "A3",
    freq: 1000
  });
  var temperature = new five.Thermometer({
    controller: "GROVE",
    pin: "A2",
    freq: 1000
  });

   var c = 0;

  temperature.on("data", function() {
    c = Math.round(this.celsius);
    if ( c >=24 ){
    led.on();
    led1.off();
   lcd.cursor(0,5).print("HOT ");
    lcd.bgColor("red");
    publish("Bridge","cupOUT",new Buffer(c+"*c"+" HOT"))
  } else if ( c <=20){
   led.off();
   led1.on();
  lcd.cursor(0,5).print("Cold");
    lcd.bgColor("blue");
    publish("Bridge","cupOUT",new Buffer(c+"*c"+"COLD"))
  } else {
   led.off();
   led1.off();
  lcd.cursor(0,5).print("Warm");
    lcd.bgColor("green");
  }
  lcd.cursor(0,0).print(c+"*c");
  });
    
  gas.on("change", function() {
      if (this.scaleTo(100,2000) > 300) {
          led0.on();
          lcd.cursor(1,0).print('AL: Beer');
          lcd.bgColor("yellow");
      } else if (this.scaleTo(100,2000) > 450) {
      led.blink(500)
      lcd.cursor(1,0).print('AL: Wine');
       lcd.bgColor("purple");
      } else {
          led0.off()
          lcd.cursor(1,0).print('Normal        ');
      }
  });
    this.loop(500,function() {
        
    
    if (cmdflag ==1) {
        lcd.cursor(0,10).print("freeze");
        
    } else {
        lcd.cursor(0,10).print("warm  ");
       
    }
    }); 
});
start();

dtex, 2016-08-25:
@chaos39 It's not clear which program you want to shut off. You have three systems:

1) Your Edison/Arduino block combo that is attached to an LCD, some LED's and a couple of sensors
2) A "server"
3) Your phone

Are you wanting to "shut off" the application running on the Edison? What do you mean by "shut off"? I ask because something still needs to be running (listening) for you to be able to turn it back on later.

chaos39, 2016-08-25:
Hi, dtex, I do apologize for the lack of information , I want to shut off the LCD, the sensors, and those leds but I want to able turn it back on later

chaos39, 2016-08-26:
So Sorry for the confusion, what I meant by shut-off was. temporally off the program such as my sensors and LCD, but when I prompt  case"ON": from my phone, I should be able to switch on my program again to fetch those readings

dtex, 2016-08-26:
LCD and LED both have ```off()``` methods and you can choose to ignore sensor data when a certain condition that you define is met i.e. ```isOff === true```. The off state of LED is very "off". No voltage is being consumed. LCD might be different, I don't really know as I don't work with LCD, but it sure does look to a casual observer like everything is off.

That was the easy answer. The hard answer involves switching power to the LCD and sensors and brings up all sorts of problems with LCD initialization and managing pin state. I really don't think it's a practical solution for you.

chaos39, 2016-08-26:
Hi Dtex, thanks for the fast reply. 
Upon understanding from what you had written. it seems that it is not very practical for me to continue this method. I shall find another way to handle this kind of problem.

Thanks for all the replies whenever I have problems with programming stuff

closed 2016-08-26 by chaos39