rwaldron/johnny-five #1210: Sweep not logging callback function

kirill2485, 2016-08-27:
I am using a servo and I am trying to log "sweep" to the console every time it does a sweep, however, it never logs it.

Here is my code:

```
var five = require('johnny-five')  
    var board = new five.Board()  
    console.log("done");
    board.on('ready', function () {  
        var servo = new five.Servo(9);
        servo.sweep().on("sweep:full", function(){
            console.log("done");
        })

    })  

```
and here is my output

```
Kirills-MacBook-Pro:bots kirillshilov$ node index2.js 
1472334792433 Device(s) /dev/cu.usbmodem1411  
1472334792443 Connected /dev/cu.usbmodem1411  
1472334795760 Repl Initialized  
\>> 

```

It does the sweeps, but it never outputs "sweep" to the console

dtex, 2016-08-27:
@kirill2485 The only event emitted by servo is "move:complete". Did you see "sweep:full" documented somewhere?

kirill2485, 2016-08-27:
@dtex http://johnny-five.io/examples/servo-sweep/

"servo.sweep().on("sweep:full", function() {"

dtex, 2016-08-27:
Thanks. Let me investigate and see when it was removed and why.

dtex, 2016-08-28:
It got lost in #858... No explanation in the PR thread, but I see it was never in the wiki either so I probably just overlooked it... I'll get it back in there if I can or change the example if I can't.

rwaldron, 2016-08-29:
I'd like to get the coverage for lib/servo.js to 100% before we address this regression. I'll that today

closed 2019-01-07 by stale[bot]