rwaldron/johnny-five #1211: How to use Johnny-Five with Cluster?

overflowz, 2016-08-28:
Well, hello there! I'd like to know how can I use johnny-five with cluster module of NodeJS?
Reason is, sometimes I get a lot uncaughtExceptions from the library and I'm unable to handle them properly (e.g if sensor dies on a pin (DS18B20) it throws up error device not found).

Second, I added DHT11 feature in ConfigurableFirmata sketch (https://github.com/taoyuan/Firmata/tree/master/examples/DHT11Firmata). For personal reason, I can't use additional I2C devices for each sensor. So when it dies (or disconnected, whatever) the library does a lot of loop conditions (loopCnt = 10000) which causes ConfigurableFirmata to freeze, thus johnny-five can't work with it too for a while (in my case, I'm reading in loop DS18B20 sensor data, but when sketch is 'frozen', it throws up uncaughtException and asking if I'm using OneWire on ConfigurableFirmata).

I also thought about domains, but sadly it's deprecated and getting removed anytime soon.

Any thoughts?

lyzadanger, 2016-08-31:
Hi, @overflowz ! This sounds like some interesting directions to go, but you're suggesting some research-and-development-level stuff that hasn't been explored by the Johnny-Five maintainers—that is, you're combining some pieces that haven't been combined before, so your knowledge is as good as any of ours. Is there a more specific question we could help answer (versus generally making something as-yet-untried work)?

overflowz, 2016-08-31:
Actually not and it's not necessary too (one worker would be enough, for sure). Would explain why, but my English skills ain't that perfect. It's just not necessary.

Thanks for the response, would be great to delete this topic, cause it makes no sense (yet?)

closed 2016-08-31 by overflowz