rwaldron/johnny-five #1212: Device or Firmware Error

filyps98, 2016-08-30:
Hi everyone, I'm a new member of github. I've been using Arduino for a year now and I wanted to create a server with the Pi and the UNO. I encountered so many problems maybe because I've bought an OrangePi which is a clone of Raspberry Pi (Yes, I know, i should have bought the original one, but i had to save some money). I' ve installed Debian 8 on it and in order to work with node I also installed  NVM and  I'm currently working with node 6. I installed johnny-five, but it gave me errors regarding serialport; to solve this problem I decided to install serialport without installing johnny-five, but there were errors about node-pre-gyp. So I installed it globally. I started the program called Hello World after uploading StandardFirmata on the arduino (not original either) and after plugging it in one of the  USB ports. The program couldn't start because the cmd was continuously saying "board looking for connected device", the next thing  that I did was looking for the different devices attached to the board and this happened:  
                                                                                                                                                                       `[ { comName: '/dev/ttyS0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/ttyS2',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/ttyS3',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined } ]`

After struggling so much I decided to change the code by defining the port . 
I tried all the three ports and the cmd always said:  

`1472576273960 Connected /dev/ttyS3
1472576283987 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:163:17)
    at Board.log (/root/node_modules/johnny-five/lib/board.js:625:8)
    at Board.(anonymous function) [as error] (/root/node_modules/johnny-five/lib/board.js:636:14)
    at Board.<anonymous> (/root/node_modules/johnny-five/lib/board.js:385:14)
    at tryOnTimeout (timers.js:232:11)
    at Timer.listOnTimeout (timers.js:202:5)`

Thank you for the help. :)

lyzadanger, 2016-08-31:
Hi! There are a number of factors at one time here, so let's break it down.

First, `node-serialport`. Getting this to compile and install successfully is key. When you say:

> So I installed it globally

Does that mean you were successful in compiling and installing `node-serialport` _globally_? What's your global version of Node.js, if so? Is it the same as the version in your local project? Can you provide me the exact version.

I'm not up to speed on `node-serialport`'s support on Debian for whatever architecture your OrangePi is on (it looks like OrangePi has an ARM Cortex V7, which is comparable to a "real" Pi 2 B).

In any case, let's start with: can you provide the exact errors you get when you try to install `johnny-five` (or `node-serialport` directly) and your exact OS and Node.js versions? Thanks!

filyps98, 2016-09-01:
Sorry, but currently I can't connect to the internet with the Pi via ethernet, so I can't provide the errors. If I remember correctly, I installed node v 6.1. something   and I only typed `npm install serialport`, but it gave me errors about node-pre-gyp (It wasn't able to install it). I decided to install this module by myself and I did that by typing `npm install node-pre-gyp -g` . I'll give more information when I manage to solve the connection problem.

lyzadanger, 2016-09-01:
@filyps98 Sounds like you've got a pile of struggles with the Pi beyond the Johnny-Five (and `node-serialport` dependency) problems you're running into. Do you want to reconnect once you've got the Pi more stable and connected and can provide error output?

Have you tried running raspbian on that Pi? I presume you have a reason to run Debian but am curious.

filyps98, 2016-09-01:
Yeah, I tried with raspbian, but it wasn't the official version, it was adapted for the orange (don't know if it changes anything) and I wasn't able to install a version of node higher than 0.10 . I'll try another time with debian and then I'll change to raspbian. Debian is behaving in a really strange way right now. Thank you.

rwaldron, 2016-09-01:
@reconbot @beriberikix ping

beriberikix, 2016-09-01:
I haven't used Orange Pis myself but they're notorious for having poor driver support and other wonkiness due to out of date OS images. First - which board and exact image did you use? Have you looked at [Armbian](http://www.armbian.com/)?

Another trick I try for new devboards is try to get the Arduino IDE working. There's usually workarounds to find online and a good litmus test for the Serial driver.

filyps98, 2016-09-01:
I have an orange pi PC, the OS that I've tried are: Debian Jessie minimal and Raspbian (which has Arduino IDE installed). I've never installed Armbian. Should I try to make Arduino work with this old version of raspbian and with the Arduino IDE before working with node? Should I install armbian after having tried?

beriberikix, 2016-09-01:
If you can program an arduino from either build, that means serial is working. If so, Armbian may not be necessary. Did you try flashing using the IDE?

filyps98, 2016-09-01:
I tried with raspbian, but it doesn't work because the OS doesn't have USBtoSerial. I've only used the terminal with Debian Jessie (which does have it) and I didn't manage to switch to the desktop,  so I wasn't able to run or download Arduino. Moreover I can't succeed in accessing via ssh the pi now, it is like the pi doesn't connect to my network...i don't know if this is a problem of firmware or hardware. So much struggle.

beriberikix, 2016-09-01:
Hmm...the latest Raspbian definitely has support for usb to serial and Arduinos. It also has `serial` support, but that may need to be enabled via [`raspi-config`](https://www.raspberrypi.org/documentation/configuration/raspi-config.md). It is hard to tell what is enabled because it looks like the fork [hosted by Orange Pi](http://www.orangepi.org/downloadresources/) for the Orange Pi PC is very old. You can run Arduino with the [official CLI](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) without the need for the desktop.

Sounds frustrating! Do you happen to have a [TTL cable](https://www.adafruit.com/products/954)? That would get around your networking issues.

filyps98, 2016-09-01:
The orange version of raspbian is really bad. From what I read in the orangepi forum, some users complain about the fact that this OS is totally different from the real one and it doesn't have some important features, so I opted for Linux which is said to be more complete. I tried to connect the Pi to the router with Ethernet in order to get internet (which is necessary for installing buildessential and node) and IT WORKED...for a while. Unfortunately not anymore, the Ethernet port of the pi seems working and the router recognizes the device, but my main computer doesn't see its ip and consequently I cannot work via ssh.

filyps98, 2016-09-01:
When I manage to access internet, I'll surely try to install Arduino. However thanks for helping me very much!!!

filyps98, 2016-09-06:
I managed to solve all the problems that I had except for Johnny-Five. I installed node v6.5.0 and npm v3.10.3. These is what i did and the output.

```
root@OrangePI:~# mkdir nodebot && cd nodebot;
root@OrangePI:~/nodebot# npm install johnny-five;

> serialport@4.0.1 install /root/nodebot/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-node-v48-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and node@6.5.0 (node-v48 ABI) (falling back to source compile with node-gyp)
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
gyp WARN download NVM_NODEJS_ORG_MIRROR is deprecated and will be removed in node-gyp v4, please use NODEJS_ORG_MIRROR
make: Entering directory '/root/nodebot/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
  CXX(target) Release/obj.target/serialport/src/serialport_unix.o
  CXX(target) Release/obj.target/serialport/src/serialport_poller.o
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/root/nodebot/node_modules/serialport/build'
/root/nodebot
└─┬ johnny-five@0.10.0
  ├── browser-serialport@2.0.3
  ├─┬ chalk@1.1.3
  │ ├── ansi-styles@2.2.1
  │ ├── escape-string-regexp@1.0.5
  │ ├─┬ has-ansi@2.0.0
  │ │ └── ansi-regex@2.0.0
  │ ├── strip-ansi@3.0.1
  │ └── supports-color@2.0.0
  ├── color-convert@1.2.2
  ├── ease-component@1.0.0
  ├── es6-shim@0.35.1
  ├── firmata@0.14.1
  ├── lodash.clonedeep@4.5.0
  ├── lodash.debounce@4.0.8
  ├── nanotimer@0.3.10
  ├─┬ serialport@4.0.1
  │ ├── bindings@1.2.1
  │ ├─┬ commander@2.9.0
  │ │ └── graceful-readlink@1.0.1
  │ ├─┬ debug@2.2.0
  │ │ └── ms@0.7.1
  │ ├── es6-promise@3.2.1
  │ ├── nan@2.4.0
  │ ├─┬ node-pre-gyp@0.6.29
  │ │ ├─┬ mkdirp@0.5.1
  │ │ │ └── minimist@0.0.8
  │ │ ├─┬ nopt@3.0.6
  │ │ │ └── abbrev@1.0.9
  │ │ ├─┬ npmlog@3.1.2
  │ │ │ ├─┬ are-we-there-yet@1.1.2
  │ │ │ │ ├── delegates@1.0.0
  │ │ │ │ └─┬ readable-stream@2.1.4
  │ │ │ │   ├── buffer-shims@1.0.0
  │ │ │ │   ├── core-util-is@1.0.2
  │ │ │ │   ├── inherits@2.0.1
  │ │ │ │   ├── isarray@1.0.0
  │ │ │ │   ├── process-nextick-args@1.0.7
  │ │ │ │   ├── string_decoder@0.10.31
  │ │ │ │   └── util-deprecate@1.0.2
  │ │ │ ├── console-control-strings@1.1.0
  │ │ │ ├─┬ gauge@2.6.0
  │ │ │ │ ├── aproba@1.0.4
  │ │ │ │ ├── has-color@0.1.7
  │ │ │ │ ├── has-unicode@2.0.1
  │ │ │ │ ├── object-assign@4.1.0
  │ │ │ │ ├── signal-exit@3.0.0
  │ │ │ │ ├─┬ string-width@1.0.1
[... 107 lines cut ...]
  │ │   │     └── concat-map@0.0.1
  │ │   ├─┬ once@1.3.3
  │ │   │ └── wrappy@1.0.2
  │ │   ├─┬ readable-stream@2.1.4
  │ │   │ ├── buffer-shims@1.0.0
  │ │   │ ├── core-util-is@1.0.2
  │ │   │ ├── inherits@2.0.1
  │ │   │ ├── isarray@1.0.0
  │ │   │ ├── process-nextick-args@1.0.7
  │ │   │ ├── string_decoder@0.10.31
  │ │   │ └── util-deprecate@1.0.2
  │ │   └── uid-number@0.0.6
  │ └─┬ object.assign@4.0.4
  │   ├─┬ define-properties@1.1.2
  │   │ └── foreach@2.0.5
  │   ├── function-bind@1.1.0
  │   └── object-keys@1.0.11
  └── temporal@0.5.0

npm WARN enoent ENOENT: no such file or directory, open '/root/nodebot/package.json'
npm WARN nodebot No description
npm WARN nodebot No repository field.
npm WARN nodebot No README data
npm WARN nodebot No license field.
```

reconbot, 2016-09-13:
@filyps98 that looks like it installed successfully

rwaldron, 2016-09-13:
👍

dtex, 2018-01-05:
@filyps98 Please feel free to reopen if you are still having an issue.

closed 2018-01-05 by dtex