rwaldron/johnny-five #1213: DHT11 connected to GrovePi+

rickysullivan, 2016-09-01:
I can't seem to get my app running:

```
var raspi = require('raspi-io');
var five = require('johnny-five');

var board = new five.Board({
    io: new raspi()
});

board.on("ready", function() {

  var temperature = new five.Thermometer({
    controller: "GROVEPI",
    pin: "A0"
  });

  temperature.on("data", function() {
    console.log("celsius: %d", this.C);
    console.log("fahrenheit: %d", this.F);
    console.log("kelvin: %d", this.K);
  });
  
});

 I've had success with the LCD, but none of the other sensors.

rwaldron, 2016-09-01:
Thanks for filing! The DHT11 sensor requires an intermediary μC that we've dubbed a "backpack", inspired by Adafruit's products of a similar nature. Take a look at this example: http://johnny-five.io/examples/multi-DHT11_I2C_NANO_BACKPACK/

rickysullivan, 2016-09-01:
Still getting error:

```
events.js:141
      throw er; // Unhandled 'error' event
      ^

Error: EIO: i/o error, read
    at Error (native)

I've plugged the DHT11 into I2C-1.

rwaldron, 2016-09-02:
I' not sure I understand what you mean by this: 

> I've plugged the DHT11 into I2C-1.

The DHT11 itself isn't an I2C device—you'd need to connect it to another microcontroller that itself has the provided firmware running on it: 

[image]

Then, your code should be: 

```js
var temp = new five.Thermometer({
  controller: "DHT11_I2C_NANO_BACKPACK"
});

temp.on("change", function() {
  console.log("temperature");
  console.log("  celsius           : ", this.celsius);
  console.log("  fahrenheit        : ", this.fahrenheit);
  console.log("  kelvin            : ", this.kelvin);
  console.log("--------------------------------------");
});
```

The controller wouldn't be a `GROVEPI`, since the Grove Pi itself is just another I2C device on the bus. The controller is our own `DHT11_I2C_NANO_BACKPACK`, which knows how to speak to the provided firmware, which you upload to whatever intermediary microcontroller you choose to use (in the diagram, I've used a nano, which I highly recommend)

rwaldron, 2016-10-14:
Have you had  a chance to try the example shown above?

dtex, 2018-01-16:
Hi @rickysullivan 

I'm going to close this since we haven't heard back. Feel free to re-open if necessary.

closed 2018-01-16 by dtex