rwaldron/johnny-five #1216: DHT21 Support ?

gedeagas, 2016-09-06:
Hi, i was wondering if the DHT21 Humidity sensor will be supported anytime soon?
or this is something that impossible because of firmata limitation ?

rwaldron, 2016-09-06:
It could be supported, in the same way shown: 

- https://github.com/rwaldron/johnny-five/issues/1213#issuecomment-244423615
- http://johnny-five.io/examples/temperature-DHT11_I2C_NANO_BACKPACK/

rwaldron, 2016-09-06:
I've just ordered one of these and will report back when it's arrived and I have a chance to work with it

gedeagas, 2016-09-09:
Sweet. 
in meantime i will report back if my DHT21 works

rwaldron, 2016-09-10:
Looks like my sensor was delivered, but I'm out of town until tomorrow night so I'll be able to look at this on Monday. Again, I will continue to keep you posted 😀

rwaldron, 2016-09-10:
What's wrong issue, I confused this with the DHT22 issue

rwaldron, 2016-09-26:
Ok, to clear this up: according to amazon I did order this sensor, but I have no idea where it actually is. I've ordered another just now

gedeagas, 2016-09-29:
Okay, i think i can test it out in the next few days
will update soon

closed 2016-10-14 by rwaldron