rwaldron/johnny-five #1217: how it works that the JS code coulb be burned into an arduino UNO?

cyaolong, 2016-09-06:
Question same as the title, don't quite understand how it works.
Someone help me the newbie?

dtex, 2016-09-06:
The JS code does not get written to the Arduino. There is a program for the Arduino called "firmata". It's a pretty standard thing and even comes pre-installed on many Arduinos.

Pasted from the Firmata repo

>Firmata is a protocol for communicating with microcontrollers from software on a host computer. The protocol can be implemented in firmware on any microcontroller architecture as well as software on any host computer software package. 

Firmata is not just a javascript thing. There are firmata libraries for JS, Python, Processing and probably others too. I recall seeing stuff done in scratch, Go, Rust and more. 

Does that help?

rwaldron, 2016-09-06:
Thanks for all the participation <3

closed 2016-09-06 by rwaldron

cyaolong, 2016-09-07:
@dtex Thanks a lot!
One more question, now that we know we could control boards through Firmata and JS with the APIs, not burning the code into the boards, but how do we connect and control the boards by using JS code? I don't quite understand what the intermediary is. Is it a web console established on the localhost?

cyaolong, 2016-09-07:
@dtex 
https://github.com/firmata/arduino
I read the firmata Usage again and I figure it out finally. Thanks again!