rwaldron/johnny-five #1219: install error WIN7 node js

alaprajz, 2016-09-08:
0 info it worked if it ends with ok
1 verbose cli [ 'C:\\Program Files\\nodejs\\node.exe',
1 verbose cli   'C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js',
1 verbose cli   'install',
1 verbose cli   'johnny-five' ]
2 info using npm@3.10.3
3 info using node@v6.5.0
4 silly loadCurrentTree Starting
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 silly fetchPackageMetaData johnny-five
8 silly fetchNamedPackageData johnny-five
9 silly mapToRegistry name johnny-five
10 silly mapToRegistry using default registry
11 silly mapToRegistry registry https://registry.npmjs.org/
12 silly mapToRegistry data Result {
12 silly mapToRegistry   raw: 'johnny-five',
12 silly mapToRegistry   scope: null,
12 silly mapToRegistry   escapedName: 'johnny-five',
12 silly mapToRegistry   name: 'johnny-five',
12 silly mapToRegistry   rawSpec: '',
12 silly mapToRegistry   spec: 'latest',
12 silly mapToRegistry   type: 'tag' }
13 silly mapToRegistry uri https://registry.npmjs.org/johnny-five
14 verbose request uri https://registry.npmjs.org/johnny-five
15 verbose request no auth needed
16 info attempt registry request try #1 at 14:11:27
17 verbose request id 9532b3e6cb91a658
18 verbose etag "2M8NOL81HWBNL2TKWT0QN9Y4Y"
19 http request GET https://registry.npmjs.org/johnny-five
20 http 304 https://registry.npmjs.org/johnny-five
21 verbose headers { date: 'Thu, 08 Sep 2016 12:11:29 GMT',
21 verbose headers   via: '1.1 varnish',
21 verbose headers   'cache-control': 'max-age=300',
21 verbose headers   etag: '"2M8NOL81HWBNL2TKWT0QN9Y4Y"',
21 verbose headers   age: '0',
21 verbose headers   connection: 'keep-alive',
21 verbose headers   'x-served-by': 'cache-fra1235-FRA',
21 verbose headers   'x-cache': 'HIT',
21 verbose headers   'x-cache-hits': '1',
21 verbose headers   'x-timer': 'S1473336689.811524,VS0,VE746',
21 verbose headers   vary: 'Accept-Encoding' }
22 silly get cb [ 304,
22 silly get   { date: 'Thu, 08 Sep 2016 12:11:29 GMT',
22 silly get     via: '1.1 varnish',
22 silly get     'cache-control': 'max-age=300',
22 silly get     etag: '"2M8NOL81HWBNL2TKWT0QN9Y4Y"',
22 silly get     age: '0',
22 silly get     connection: 'keep-alive',
22 silly get     'x-served-by': 'cache-fra1235-FRA',
22 silly get     'x-cache': 'HIT',
22 silly get     'x-cache-hits': '1',
22 silly get     'x-timer': 'S1473336689.811524,VS0,VE746',
22 silly get     vary: 'Accept-Encoding' } ]
23 verbose etag https://registry.npmjs.org/johnny-five from cache
24 verbose get saving johnny-five to C:\Users\User\AppData\Roaming\npm-cache\registry.npmjs.org\johnny-five\.cache.json
25 verbose correctMkdir C:\Users\User\AppData\Roaming\npm-cache correctMkdir not in flight; initializing
26 silly install normalizeTree
27 silly loadCurrentTree Finishing
28 silly loadIdealTree Starting
29 silly install loadIdealTree
30 silly cloneCurrentTree Starting
31 silly install cloneCurrentTreeToIdealTree
32 silly cloneCurrentTree Finishing
33 silly loadShrinkwrap Starting
34 silly install loadShrinkwrap
35 silly loadShrinkwrap Finishing
36 silly loadAllDepsIntoIdealTree Starting
37 silly install loadAllDepsIntoIdealTree
38 silly rollbackFailedOptional Starting
39 silly rollbackFailedOptional Finishing
40 silly runTopLevelLifecycles Starting
41 silly runTopLevelLifecycles Finishing
42 silly install printInstalled
43 verbose stack Error: Refusing to install johnny-five as a dependency of itself
43 verbose stack     at checkSelf (C:\Program Files\nodejs\node_modules\npm\lib\install\validate-args.js:53:14)
43 verbose stack     at Array.<anonymous> (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\bind-actor.js:15:8)
43 verbose stack     at LOOP (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\chain.js:15:14)
43 verbose stack     at chain (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\chain.js:20:5)
43 verbose stack     at C:\Program Files\nodejs\node_modules\npm\lib\install\validate-args.js:16:5
43 verbose stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\async-map.js:52:35
43 verbose stack     at Array.forEach (native)
43 verbose stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\async-map.js:52:11
43 verbose stack     at Array.forEach (native)
43 verbose stack     at asyncMap (C:\Program Files\nodejs\node_modules\npm\node_modules\slide\lib\async-map.js:51:8)
44 verbose cwd C:\
45 error Windows_NT 6.1.7601
46 error argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "johnny-five"
47 error node v6.5.0
48 error npm  v3.10.3
49 error code ENOSELF
50 error Refusing to install johnny-five as a dependency of itself
51 error If you need help, you may report this error at:
51 error     <https://github.com/npm/npm/issues>
52 verbose exit [ 1, true ]

rwaldron, 2016-09-08:
Do you have a package.json in this directory that contains: 

```
  "name": "johnny-five",
```

?

alaprajz, 2016-09-08:
Thanks for quick answer.
Where could I download that from?

rwaldron, 2016-09-08:
I don't understand what you mean, download what? I'm asking if the directory that you're attempting to install in has a package.json file, and if it contains a "name" key whose value is "johnny-five"

alaprajz, 2016-09-08:
Yes, it contains that in first line.

rwaldron, 2016-09-08:
That's causing this error. Change the value "johnny-five" to anything else and npm will stop failing.

alaprajz, 2016-09-08:
Answer:
Invalid name
and 10 warning of new name
No description
No respository field
No Readme file
No licence field

Looking for connected device

rwaldron, 2016-09-08:
I think you need to start from a fresh project directory, because I have no idea what's going on. If you saw "Looking for connected device" in your terminal, that means johnny-five is definitely installed. Were you maybe trying to install johnny-five again from within the johnny-five module itself? 

1. Create a new directory called "test-install-j5" in your $HOME directory
2. Navigate into "test-install-j5"
3. Run: "npm init -y"
4. Run: "npm install johnny-five"
5. Report back the _exact_ output of doing that

alaprajz, 2016-09-09:
C:\>cd test-install-j5

C:\test-install-j5>npm init -y
Wrote to C:\test-install-j5\package.json:

{
  "name": "test-install-j5",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
  },
  "keywords": [],
  "author": "",
  "license": "ISC"
}

C:\test-install-j5>npm install johnny-five

> serialport@4.0.1 install C:\test-install-j5\node_modules\serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success:
"C:\test-install-j5\node_modules\serialport\build\Release\
serialport.node" is installed via remote
test-install-j5@1.0.0 C:\test-install-j5
`-- johnny-five@0.10.0
  +-- browser-serialport@2.0.3
  +-- chalk@1.1.3
  | +-- ansi-styles@2.2.1
  | +-- escape-string-regexp@1.0.5
  | +-- has-ansi@2.0.0
  | | `-- ansi-regex@2.0.0
  | +-- strip-ansi@3.0.1
  | `-- supports-color@2.0.0
  +-- color-convert@1.2.2
  +-- ease-component@1.0.0
  +-- es6-shim@0.35.1
  +-- firmata@0.14.1
  +-- lodash.clonedeep@4.5.0
  +-- lodash.debounce@4.0.8
  +-- nanotimer@0.3.10
  +-- serialport@4.0.1
  | +-- bindings@1.2.1
  | +-- commander@2.9.0
  | | `-- graceful-readlink@1.0.1
  | +-- debug@2.2.0
  | | `-- ms@0.7.1
  | +-- es6-promise@3.2.1
  | +-- nan@2.4.0
  | +-- node-pre-gyp@0.6.29
  | | +-- mkdirp@0.5.1
  | | | `-- minimist@0.0.8
  | | +-- nopt@3.0.6
  | | | `-- abbrev@1.0.9
  | | +-- npmlog@3.1.2
  | | | +-- are-we-there-yet@1.1.2
  | | | | +-- delegates@1.0.0
  | | | | `-- readable-stream@2.1.4
  | | | |   +-- buffer-shims@1.0.0
  | | | |   +-- core-util-is@1.0.2
  | | | |   +-- inherits@2.0.1
  | | | |   +-- isarray@1.0.0
  | | | |   +-- process-nextick-args@1.0.7
  | | | |   +-- string_decoder@0.10.31
  | | | |   `-- util-deprecate@1.0.2
  | | | +-- console-control-strings@1.1.0
  | | | +-- gauge@2.6.0
  | | | | +-- aproba@1.0.4
[... 110 lines cut ...]
  | |   |   `-- brace-expansion@1.1.5
  | |   |     +-- balanced-match@0.4.1
  | |   |     `-- concat-map@0.0.1
  | |   +-- once@1.3.3
  | |   | `-- wrappy@1.0.2
  | |   +-- readable-stream@2.1.4
  | |   | +-- buffer-shims@1.0.0
  | |   | +-- core-util-is@1.0.2
  | |   | +-- inherits@2.0.1
  | |   | +-- isarray@1.0.0
  | |   | +-- process-nextick-args@1.0.7
  | |   | +-- string_decoder@0.10.31
  | |   | `-- util-deprecate@1.0.2
  | |   `-- uid-number@0.0.6
  | `-- object.assign@4.0.4
  |   +-- define-properties@1.1.2
  |   | `-- foreach@2.0.5
  |   +-- function-bind@1.1.0
  |   `-- object-keys@1.0.11
  `-- temporal@0.5.0

npm WARN test-install-j5@1.0.0 No description
npm WARN test-install-j5@1.0.0 No repository field.

C:\test-install-j5>

rwaldron, 2016-09-12:
Sorry, I was away for a few days on a trip with my wife. By the looks of that output, everything should be installed exactly as expected. Have you tried writing any J5 programs?

alaprajz, 2016-09-12:
Thanks a lot, I hope that your trip has been spent greatfully.
I couldn't use arduino board with J5 unfortunatelly.
Have you got idea for check the COM ports?

rwaldron, 2016-09-12:
Johnny-Five should just automatically detect COM ports with arduino's attached. Sometimes windows machines will have COM ports that aren't attached to anything. You'll need to plug in your arduino and open windows device manager and figure out if there are any COM ports with lower numbers than the one appearing for the arduino. Ignore those lower ones by [specifying the port](http://johnny-five.io/api/board/)

alaprajz, 2016-09-13:
avrdude: ser_open(): can't open device "\\.\COM1":

alaprajz, 2016-09-13:
The solution is:
needed setup FTDI 2.8.2.0
and
needed setup CDM v2.08.28 Certified (refrest tool from downloaded folder in
device/tool manager)
COM3 has been added
it works!

2016-09-13 18:00 GMT+02:00 Tibor Csala <csalatibor@gmail.com>:

rwaldron, 2016-09-13:
Great!

closed 2016-09-13 by rwaldron