rwaldron/johnny-five #1220: Multi arduino issue

drj-io, 2016-09-11:
### Board: 
2x Arduino Mega boards
### Shield:
Spark fun big easy driver
https://www.sparkfun.com/products/12859

### Hardware you are having an issue with: 
Steppers

### Version of Johnny-Five: 
Tested with 0.10.0

### What your expectations are: 
Trying to use 2 different boards with Johnny Five, but it always uses the same board.

### What the actual outcome is: 
It always uses the same board, usually the first one that was plugged in.

Steps to reproduce (including code samples):
```javascript
var serialport = require('serialport');
var five = require('johnny-five');

serialport.list(function(err,ports){
    var connected_boards = [];
    console.log(ports);
   var ii = 0;
    ports.forEach(function(port,i){
      ii++;
      if(port.manufacturer == 'Arduino (www.arduino.cc)'){
        connected_boards.push({id: "a"+ii, repl:false});
      }
    });
    var boards = new five.Boards(connected_boards);
    boards.on("ready", function() {
      this.each(function(board){
        console.log("-->",board.io.firmware.name)

        if(board.io.firmware.name=="blue.ino"){
          var stepper = new five.Stepper({
            type: five.Stepper.TYPE.DRIVER,
            stepsPerRev: 3200,
            pins: [23,22]
          });
          stepper.rpm(180).ccw().step(6200, function(){
            console.log("ok")
          });
        }
      })
    });
  })
```

Using this code, I have 2 arduino's attached (firmware "blue.no" and "red.ino" respectively).  When I run it the first time "red.ino" turns the stepper attached to it.  If I change "blue.ino" to "red.ino", red.ino turns the stepper on the same port.

I can get this to flip by unplugging both devices and plugging them back in in a different order.

rwaldron, 2016-09-11:
All component instantistions must have the `board` property explicitly set:

```Js   
         var stepper = new five.Stepper({
            board: board, // <--- as named in your code above 
            type: five.Stepper.TYPE.DRIVER,
            stepsPerRev: 3200,
            pins: [23,22]
          });
``` 

... Otherwise J5 defaults to the first board created and stored in its internal cache. This is only necessary when using multiple boards.

closed 2016-09-11 by rwaldron

drj-io, 2016-09-11:
Easy thanks!

rwaldron, 2016-09-11:
No problem! :)