rwaldron/johnny-five #1222: pinMode async?

ugate, 2016-09-12:
Is there any reason why `pinMode` isn't *async*? It would seem as though there would be a need in some cases to perform *async* operations (e.g. `fs.write`, etc.) before continuation of pin operations (e.g. `analogWrite`, `digitalWrite`, etc.).

```js
// test
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", () => {
  this.pinMode(9, five.Pin.PWM);
  this.analogWrite(9, 255);
});

// in BoardIO impl
function pinMode(pin, mode) {
  ...
  // only alternative is to use sync alternative
  fs.write(fd, value, 0, value.length, err => {
    if (err) throw err; // analogRead has already been executed!!!
    ...
  });
  ...
}
```

An *async* approach:

```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", () => {
  this.pinMode(9, five.Pin.PWM, mode => {
    this.analogWrite(9, 255);
  });
});

// in BoardIO impl
function pinMode(pin, mode, handler) {
  ...
  fs.write(fd, value, 0, value.length, err => {
    if (err) throw err;
   handler(mode); // now we can let analogWrite execute
    ...
  });
  ...
}
```

rwaldron, 2016-09-12:
Write operations should be thought of as "sent", it's up to the underlying [IO plugin](https://github.com/rwaldron/io-plugins) to handle them in the most optimized way possible for whatever platform the code is running on or interacting with. Writing to sysfs generally means: open and write a single byte—"async" writing often imposes artificial latency between operations, which is undesirable. Additionally, no platforms that I've encountered include any signaling that the write is "complete"—either it completed correctly or it didn't. Furthermore, some program may often want to send several write messages back to back, as quickly as possible and absolutely in the same execution turn. Forcing writes to be async would prevent that. 

What platform are you working with? I suggest that you spend some time with the platforms that are supported, that way you'll get a better understanding for how the lower level operations behave. http://johnny-five.io/platform-support/

closed 2016-09-12 by rwaldron

ugate, 2016-09-13:
@rwaldron Thank you for the detailed explanation. The use of `fs` was simply an example use case for *async*. Although, there does seem to be IO plugins that use `sysfs` for *write* operations. [raspi-io](https://github.com/nebrius/raspi-io/blob/master/lib/index.js#L17) looks to be using [raspi-gpio](https://github.com/nebrius/raspi-gpio/blob/master/src/write.cc#L27) which in turn uses *WiringPi* which [uses sysfs to perform writes](https://github.com/WiringPi/WiringPi/blob/master/gpio/gpio.c). The only way I've seen to get around it is via a complex driver APIs using `nan` or writing directly to `/dev/mem` (which is okay by me, but typically frowned upon by the Linux deities). What is the recommended approach?

I've generally found that most async operations have a minimal impact on performance when you factor in blocking synchronous I/O latency which I assume a lot of applications make use of when exposing a web UI.

rwaldron, 2016-09-13:
> What is the recommended approach?

Honestly? W/R to hardware, nodebots and so forth: no one approach is ever the best for all platforms. This is both interesting and challenging, but I always recommend to the authors of those IO Plugins to use whatever they can use that the platform gives them that provides the best performance.