rwaldron/johnny-five #1224: PCF8574T lcd 16x2 backpack nothing shown in display

sumarouno, 2016-09-17:
I tried this code:
```
var five = require("johnny-five");
var chipio = require('chip-io');

var board = new five.Board({
  io: new chipio()
});

board.on("ready", function() {
  var l = new five.LCD({
        controller: "PCF8574T",
        address: 0x27,
        bus: 2,
        rows: 2,
        cols: 16
  });

  l.useChar("heart");
  l.cursor(0, 0).print("hello :heart:");
```
But nothing displayed in my lcd. thanks

lyzadanger, 2016-09-22:
@sumarouno Is your code example truncated? I just wanted to verify because it's incomplete as you quoted it (i.e. it's missing the end brackets of the `ready` callback function, etc.).

Does the code execute without error when you run it via `node`? Are you getting any output or messages?

I thought I'd try to help establish these basics before moving further. Unfortunately, I don't have an LCD backpack on hand, but there are other folks in the community who does, so we'll figure this out!

sumarouno, 2016-09-22:
@lyzadanger , Oh my bad. I'm sorry for that. Thank you for pointing me to my fault. I forgot to close my `ready` function. I close that function, and that code is running well without any error. So, this issue is solved. And I close this issue.
Thanks for replying my post. I appreciate that.

closed 2016-09-22 by sumarouno