rwaldron/johnny-five #1226: How can I make visual surface of johnny-five?

alaprajz, 2016-09-18:
How could I write visual surface with johnny-five? with javascript and html

dtex, 2016-09-18:
Hi @alaprajz 

I'm not sure what you're asking for here. Can you give us more information?

alaprajz, 2016-09-18:
I am looking for connection between johnny-five and html or visual surface.
I would like navigate my arduino in countryside from capital town.
How could I sold that?

dtex, 2016-09-19:
It sounds like you want to control and Arduino through a web browser.

Trying to use an Arduino Uno on a project like this without a host server connected to your Arduino is not an approach I would recommend so first you will want to decide if a wifi friendly Microcontroller like the [Arduino MKR1000](https://www.arduino.cc/en/Main/ArduinoMKR1000) or the [Particle Photon](https://www.particle.io/products/hardware/photon-wifi-dev-kit) is the best option for your project or if you'd be better off with a single board computer (SBC) like the [Tessel 2](http://tessel.io) or [Raspberry Pi](https://www.adafruit.com/category/105). It really depends on what you are building and both could be made to work. I believe the SBC option brings some advantages, especially if you want to use Johnny-Five.

alaprajz, 2016-09-19:
Yes, you understand my goal. Thanks, I will try Raspberry pi version.

lyzadanger, 2016-09-22:
Hi, @alaprajz ! It sounds like you're on your way to happy experimentation with a Pi. I'm going to close this issue for now, but please reopen it if we can help you with further specific questions or issues. Good luck!

closed 2016-09-22 by lyzadanger