rwaldron/johnny-five #1229: Trying lcd_runner.js but ended with TypeError: this.loop is not a function

sumarouno, 2016-09-25:
First of all, I would say thank your for the great work.
I try your code example lcd_runner.js but encountered an error.

**My hardware are**:
1. arduino mega 2560 (clone)
2. LCD 1602 with PCF8574T backpack

and here **the code** that i try:
```javascript
var five = require("johnny-five");
var board = new five.Boards(["/dev/ttyUSB0"]);

board.on("ready", function() {
	var lcd = new five.LCD({
		controller: "PCF8574T",
		address: 0x27,
		rows: 2,
		cols: 16
	});
	
	var frame = 1;
	var frames = [":runninga:", ":runningb:"];
	var col = 0;
	var row = 0;
		
	lcd.useChar("runninga");
	lcd.useChar("runningb");
	
	
	this.loop(300, function() {
		lcd.clear().cursor(row, col).print(frames[frame ^= 1]);
		
		if (++col === lcd.cols) {
			col = 0;
			if (++row === lcd.rows) {
				row = 0;
			}	
		}
	});
});
		
```

and here for the **error log**:
```bash
node mega8_lcd_runner.js 
1474775542413 Connected /dev/ttyUSB0  
1474775545637 Board ID:  FC5FBD5F-46A9-4302-807F-E85AEA6FA9CC  
1474775545643 Repl Initialized  
>> /home/sumarsono/belajar-node-red/node_modules/node-red/mega8_lcd_runner.js:21
	this.loop(300, function() {
	     ^

TypeError: this.loop is not a function
    at Boards.<anonymous> (/home/sumarsono/belajar-node-red/node_modules/node-red/mega8_lcd_runner.js:21:7)
    at emitOne (events.js:77:13)
    at Boards.emit (events.js:169:7)
    at Boards.<anonymous> (/home/sumarsono/belajar-node-red/node_modules/node-red/node_modules/johnny-five/lib/board.js:1190:14)
    at nextTickCallbackWith0Args (node.js:420:9)
    at process._tickDomainCallback (node.js:390:13)
```

any help? thanks

mwittig, 2016-09-25:
You're using the `Boards` class which is used to construct a "collection object containing multiple board objects". This class has different characteristics than the `Board` class, obviously.  So, I guess the `loop` function is not provided by design here.  

As you apparently don't need to handle multiple boards at the same time my suggestion is to replace line two of your code as follows. This way your code will use an instance of the `Board` object which provides the features you're requiring in your code example.

``` 
var board = new five.Board({ port: "/dev/ttyUSB0"});
```

See also:
https://github.com/rwaldron/johnny-five/wiki/Boards
https://github.com/rwaldron/johnny-five/wiki/Board

sumarouno, 2016-09-25:
Thank you, @mwittig. Yes, of course if using `board`  the lcd_runner.js will work. But I have a plan to use two Arduino board, that is why I try to use `boards` .
Any help?

Thank you.

mwittig, 2016-09-25:
> Any help?

Yes. In this case my suggestion is to use the `Boards` constructor and to ise ids for your boards. This way you can select the select the board later using the `byId` member function. I have rewritten your code accordingly and it should work, but I did not have a chance to test it. Hopefully, the code make sense for you. I am happy to help further, otherwise.

```
var five = require("johnny-five");
var boards = new five.Boards([ { id: "A", port: "/dev/ttyUSB0"} ]);
//   ^^^^ note, I have renamed board to boards as this is the collection of boards you'll setup

boards.on("ready", function() {
  // all boards are ready
  // now doing stuff for board with id "A"
  var state = 0;
  var board = boards.byId("A");
  board.pinMode(13, board.MODES.OUTPUT);
  // ^^^^ instead of "this" I am now using the board instance 

  board.loop(500, function() {
    board.digitalWrite(13, (state ^= 1));
  });
});
```

mwittig, 2016-09-25:
Btw, I have noticed the Boards API documentation does not describe the member functions which make it hard for beginners to use it.

sumarouno, 2016-09-25:
@mwittig , thanks for your help. Your explanation is great for me. I also try our code with little modification, here :
**try_1.js**
```javascript
var five = require("johnny-five");
var boards = new five.Boards([
	{ 
	id: "mega",
	port: "/dev/ttyUSB0"
	},
]);

boards.on("ready", function() { 
	var state = 0;
	var board_mega = boards.byId("mega");

	board_mega.pinMode(13, board_mega.MODES.OUTPUT);

	board_mega.loop(500, function() {
		board_mega.digitalWrite(13, (state ^= 1));
	});
});
```
the code above is work, the led is blinking. And then I try to attach my another arduino and try to give more modification from your code, here:
**try_2.js**
```javascript
var five = require("johnny-five");
var boards = new five.Boards([
	{ 
	id: "mega",
	port: "/dev/ttyUSB0"
	},
	{
	id: "uno",
	port: "/dev/ttyACM0"
	}
]);

boards.on("ready", function() { 
	var state_mega = 0;
        var state_mega = 0;
	var board_mega = boards.byId("mega");
	var board_uno = boards.byId("uno");

	board_mega.pinMode(13, board_mega.MODES.OUTPUT);
	board_uno.pinMode(13, board_uno.MODES.OUTPUT);

	board_mega.loop(500, function() {
		board_mega.digitalWrite(13, (state ^= 1));
	});

	board_uno.loop(500, function() {
		board_uno.digitalWrite(13, (state ^= 1));
	});
});
```
the **try_2.js** is working, the led in my mega and uno is blinking. 

I agree with you about this:
> the Boards API documentation does not describe the member functions which make it hard for beginners to use it

Thank you for your help

sumarouno, 2016-09-25:
I try another code, i add lcd to your code, and work well. here the code:
```javascript
//percobaan 2 arduino menggunakan ID
//pakai constructor Boards
//

var five = require("johnny-five");
var boards = new five.Boards([
	{ 
	id: "mega",
	port: "/dev/ttyUSB0"
	},
	{
	id: "uno",
	port: "/dev/ttyACM0"
	}
]);

boards.on("ready", function() {
	var lcd = new five.LCD({
		controller: "PCF8574T",
		address: 0x27,
		rows: 2,
		cols: 16
	}); 
	var state_mega = 0;
	var state_uno = 0;
	var board_mega = boards.byId("mega");
	var board_uno = boards.byId("uno");
	
	var frame = 1;
	var frames = [":runninga:", ":runningb:"];
	var col = 0;
	var row = 0;
		
	lcd.useChar("runninga");
	lcd.useChar("runningb");

	board_mega.pinMode(13, board_mega.MODES.OUTPUT);
	board_uno.pinMode(13, board_uno.MODES.OUTPUT);

	board_mega.loop(300, function() {
		lcd.clear().cursor(row, col).print(frames[frame ^= 1]);
		
		if (++col === lcd.cols) {
			col = 0;
			if (++row === lcd.rows) {
				row = 0;
			}	
		}
	});
	
	board_mega.loop(500, function() {
		board_mega.digitalWrite(13, (state_mega ^= 1));
	});

	board_uno.loop(500, function() {
		board_uno.digitalWrite(13, (state_uno ^= 1));
	});

});
```

sumarouno, 2016-09-25:
This issue is **solved**, so I want to close this issue.
**Thank you**

mwittig, 2016-09-25:
Your're welcome! Note, if you want do the same operations on two boards you can also use the boards.each() function to iterate the boards. See the following example taken from https://github.com/rwaldron/johnny-five/wiki/Boards

```
var ports = [
  { id: "A", port: "/dev/cu.usbmodem621" },
  { id: "B", port: "/dev/cu.usbmodem411" }
];

new five.Boards(ports).on("ready", function() {

  // Both "A" and "B" are initialized
  // (connected and available for communication)

  // |this| is an array-like object containing references
  // to each initialized board.
  this.each(function(board) {

    // Initialize an Led instance on pin 13 of
    // each initialized board and strobe it.
    new five.Led({ pin: 13, board: board }).strobe();
  });
});

rwaldron, 2016-09-27:
@mwittig @sumarouno thanks to both of you for working through this! Maybe we can evolve some of the discussion into docs or examples? Let's ask @lyzadanger for guidance on that

mwittig, 2016-09-27:
@rwaldron I am happy to contribute to docs like adding more examples or adding documentation for methods. Guidance on this is welcome!

lyzadanger, 2016-09-27:
@mwittig @rwaldron Happy to help ( @mwittig : I've been on a documentation streak of late—I like to help out in this way). It would be a good learning/synthesizing exercise for me because I've never used J5 with > 1 board at a time before! However, @mwittig, this is your issue and if you are _excited_ to contribute docs, go for it. Let me know. 

@rwaldron If I embark on docs, should I:

1. Open a new issue to track it; OR
2. Reopen this issue; OR
3. SOMETHING ENTIRELY DIFFERENT

(circle one)

rwaldron, 2016-09-27:
Usually I would say start with 1, but when there is an existing issue start at 2. Number 3...? No idea.

rwaldron, 2016-09-27:
I self assigned on accident. Ignore that

lyzadanger, 2016-09-27:
Note: I'm booked through until at least Thursday, but I hope to be able to peek at this this week! If not, next week. Hope that's OK.

rwaldron, 2016-09-27:
@mwittig can get started then? All you need to do is write some content in a gist to start :)

mwittig, 2016-09-28:
Sure. I'll write my stuff into a gist and than we can discuss how to integrate it into the project - be it jsdoc source code commments, wiki, both of the aforementioned, or something else :)

sumarouno, 2016-09-29:
Great

closed 2018-08-28 by stale[bot]