rwaldron/johnny-five #1230: J5 resets arduino state/output pins when it starts

drj-io, 2016-09-25:
Every time J5 starts, it resets all my relays, turns them all off then on. Is there any way to prevent this?

Can j5/firmata just leave the existing state of all outputs the way they are?  I'm using j5 on several projects and this is undesirable.  

edge case: when I come back to my car, all of my windows are rolled down.  This creates a security concern, of sorts. ;) 

This might be a feature request, or at least a support request.  Thanks a lot for your help.

sumarouno, 2016-09-25:
If we are using Arduino native code, we can save the last state to eeprom. Can j5 do that?
Thank

rwaldron, 2016-09-26:
Just copying the reply I gave in the gitter channel: 

> J5 isn't doing that, it's the standard state normalizing that occurs in standard firmata https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L771 > https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L722-L725 > https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino#L290-L295

There isn't anything J5 can do about this at its level in the stack

closed 2016-09-26 by rwaldron

drj-io, 2016-09-26:
OK, that's what I figured, thanks.

soundanalogous, 2016-09-27:
Some day this will be possible if I or someone else ever gets to this: https://github.com/firmata/arduino/issues/166.

sumarouno, 2016-09-27:
I agree