rwaldron/johnny-five #1232: Basic Motor Example as Shown is Slightly Dangerous

lyzadanger, 2016-09-29:
http://johnny-five.io/examples/motor/

This could lead to melting an Arduino; circuit should have protective diode at least and external power optimally. 

I can take this as part of the motor-example work I'm doing but wanted to make sure there weren't any dissenters.

dtex, 2016-09-29:
But isn't "Danger" your middle name? 😃 

We have a similar situation with the [servo.png](https://github.com/rwaldron/johnny-five/blob/master/docs/breadboard/servo.png) diagram where we are not showing external power. Brownouts are a pretty common experience when wired that way.

Resseguie, 2016-09-30:
There was some discussion on the servo one when it was first created. I seem to remember that Rick was leaning towards simplicity and shortest path to get something moving for new users. But it's a common problem that we should probably address somehow to relieve frustration when users move from micro servos to ones that are more powerful (and power hungry).

juventinoromero, 2017-12-26:
Hi @dtex, may I help out with this?

dtex, 2017-12-26:
@juventinoromero Absolutely! That would be very much appreciated.

We also have #1308 which is very closely related to this.

We keep all the fritzing diagrams in /docs/breadboard so there is plenty in there to crib from.

juventinoromero, 2017-12-30:
@dtex is the request to change the circuit image to one with a protective diode? If so, does it need to be drawn with fritzing?

dtex, 2017-12-30:
@juventinoromero Yes, that's exactly right we just need the diode added and we do need a fritzing drawing along with generated png. Are you familiar with fritzing? It's pretty easy to use (and free). 

I also think we should add a text block in the fritzing diagram warning users that most motors will require external power.

@reseguie The argument to keep things simple is very compelling but as @lyzadanger points out the risk of not having a diode on a directly connected motor is a cooked Arduino. The risk of a servo drawing too much current is just a brownout.

juventinoromero, 2017-12-31:
@dtex I've donde the changes. Here's the PR https://github.com/dtex/johnny-five/pull/1

juventinoromero, 2018-01-02:
@dtex here is the new PR: https://github.com/rwaldron/johnny-five/pull/1433