rwaldron/johnny-five #1234: Analog Thermometer pin conflicts with Button pin

ferandec, 2016-10-02:
Hello,

I'm new to Johnnny-Five but when I create a button connected to digital pin 5 and an analog thermometer on pin "A5", I lost the events on the digital pin.
I'm using an Arduino Uno.
In the code below when I change the analog pin from "A5" to "A3", the button events "press" and "release" fire but when I declare pin "A5" for the thermometer the button events don't fire (only the thermometer "data" event is fired). 
`	sw1 = new five.Button(5);
	thermo = new five.Thermometer({
		mode : "ANALOG",
		pin: "A5",
		freq : 1000,
		threshold : 5,
		toCelsius: function(raw) {
			var celsius = (raw * 0.00132 - 0.4) / 0.01953;
			//console.log("raw : " + raw + " -> " + celsius);
			return celsius;
		}
	});
[...]
		sw1.on("release", function() {
			console.log( "sw1 relâché" );
			client.broadcast.emit('pushbutton', {
				id : "D5",
				val : "off"
			});  
		});

		sw2.on("press", function() {
			console.log( "sw2 enfoncé" );
			client.broadcast.emit('pushbutton', {
				id : "D2",
				val : "on"
			});
		});

		 thermo.on("data", function() {
			 console.log("celsius: %d", this.C.toFixed(2));
		 });
[...]
`

Kind Regards

ferandec, 2016-10-02:
I want to precise that if I create the button after the thermometer, the program runs as expected.

rwaldron, 2016-10-02:
Let's try to figure this out together... I suspect that `Thermometer` doesn't understand what pin "A5" is, but I'm not sure why. There isn't a `mode` property in `Thermometer` constructor options, but maybe you meant "controller"? Try changing "mode" to "controller"

ferandec, 2016-10-03:
I confirm to you that changing the option name from "mode" to "controller" doesn't resolve the problem.
However, I think that `Thermometer` acts on digital pin due to a side-effect rather than a misinterpretation of the pin name since the temperature supplied on pin 'A5' from my arduino shield is well returned.
Thanks for your great job.

rwaldron, 2018-01-05:
I traced this through to Firmata.js and confirmed that the issue is not in J5. From there, I took a trip into the past and as far as I can tell this bug has always existed. I also tested digital writes to pin 5 while reading A5, but that worked without any issue. I suspect this is being caused by some code in StandardFirmata.

closed 2018-01-05 by rwaldron

rwaldron, 2018-01-05:
I just realized that I did not use an old version of StandardFirmata when I was doing these historical tests. That may be relevant