rwaldron/johnny-five #1235: autoscroll() not working

sumarouno, 2016-10-03:
I try `autoscroll()` for my mega2560 and my lcd 1602 with i2c backpack. What i got was "BleepBloop" with blinking cursor at the end.
here my code for m mega:

```javascript
var five = require('johnny-five');

var boards = new five.Boards([
{
    id: "mega",
    port: "/dev/ttyUSB3"
},
]);

boards.on('ready', function() {
    var lcd = new five.LCD({
        controller: "PCF8574T", 
        address: 0x27, 
        bus: 2, 
        rows: 2, 
        cols: 16
    });

        lcd.cursor(0, 0);
        lcd.autoscroll().print("Bloop").print("Bleep");
});
```
thank you

rwaldron, 2016-10-06:
Looks like autoscroll and cursor blink may have gotten mixed up!

sumarouno, 2016-10-07:
Have you fixed that? I will happy to try. Thanks

rwaldron, 2016-10-07:
The issue: J5's LCD wasn't setting _and_ storing the display "entry" bits

rwaldron, 2016-10-07:
I'll push the fix out today

sumarouno, 2016-10-09:
I will try as soon as possible, I have a job at other town right now. Thank you.

rwaldron, 2016-10-11:
Sorry for the delay, I've been moving my office since last Friday afternoon. After testing this again today, I'm noticing that the LCD no longer reliably resets, so I need to figure that out before I can push out a release

sumarouno, 2016-10-14:
thank you, i will waiting until you push the fix.

tawalke, 2016-10-23:
I am having a similiar issue with autoscroll() as well on the Intel Edison Arduino with the Grove LCD JHD1313M1/JHD1313M3 sensor on I2C on Grove shield using the Edison-IO.  

The test code is as follows: 
rgbLCD.autoscroll().print("Button was pressed % times", buttonPresses.toString());
and
rgbLCD.autoscroll();
rgbLCD.print("Button was pressed " + buttonPresses.toString() + " times");
and
rgbLCD.autoscroll().print("Button was ").print("pressed " + buttonPresses.toString()).print(" times");

As you can see by the picture the LCD only writes the first 16 characters only and does not scroll showing the rest of the string.
  
[image]

rwaldron, 2016-10-24:
Yes, the bug will exist on all platforms since it's the same code. 

I'm still having issues with the LCD not fully resetting.

rwaldron, 2016-10-24:
Sorry, I _did_ push the changes and released them in v0.10.3. I will have to double check the reset issues.

sumarouno, 2016-10-25:
Thank @rwaldron

tawalke, 2016-10-30:
Thanks, please keep us posted on any updates.  I did verify that am running v.0.10.3.  I am manually separating the rows now, but I look forward to any feedback you have. Thanks

tawalke, 2017-06-24:
Hi @rwaldron, I was using the Grove RGB LCD (I2C) for another project and noticed that the autoscroll() is still not working. Any thoughts?

rwaldron, 2017-06-26:
@tawalke nothing immediately. I'll have to test the hardware and try to reproduce. Can you paste code that reproduces? Thanks

rwaldron, 2017-07-05:
@tawalke can you try `npm install rwaldron/johnny-five#master`? I was able to reproduce the issue you had and ended up writing a special controller override method definition for `autoscroll`

tawalke, 2017-07-05:
Excellent, let me try now!

Sent from Windows Mail

ritesh2014, 2017-08-09:
The grove starter kit V3 LCD controller name is **JHD1313M3**. When I use this name it throws the error - 
>> ERROR: /home/root/.node_app_slot/node_modules/johnny-five/lib/lcd.js:642
        opts.pins.forEach(function(pin) {
So I revert back to the name JHD1313M1 and it works fine except - 
lcd.useChar("heart");
lcd.print(":heart:");
**does not** prints the symbol (screen is blank). 
This **works** - lcd.print("Hello"); bgcolor, blink etc works.

Also autoscroll **does not** works.

Do you have to add JHD1313M3 in the controller list (in lcd.js) and make necessary changes to make the above functions work?

vjustov, 2018-10-13:
Hi @rwaldron, I was playing around with an arduino uno and the Sparkfun LCD with the simple example and noticed that the autoscroll() is still not working. maybe I'm doing something wrong?

hugocruz, 2019-05-24:
Hei. 
I believe this is still an issue. @rwaldron was the fix pushed?

closed 2020-05-25 by stale[bot]