rwaldron/johnny-five #1236: TypeError: this.transport.write is not a function

federicolucca, 2016-10-10:
Hi,

I followed all the instructions but in one mac everything works in another i have this error. why?   

info  - socket.io started
1476083802468 Device(s) /dev/cu.usbmodem26431 
bitbucket/arduino_server/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:898
  this.transport.write(new Buffer([START_SYSEX, CAPABILITY_QUERY, END_SYSEX]));
                 ^

TypeError: this.transport.write is not a function
    at Board.queryCapabilities (/bitbucket/arduino_server/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:898:18)
    at Board.<anonymous> (/bitbucket/arduino_server/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:692:14)
    at Board.g (events.js:260:16)
    at emitNone (events.js:67:13)
    at Board.emit (events.js:166:7)
    at SYSEX_RESPONSE.(anonymous function) (/bitbucket/arduino_server/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:186:9)
    at Board.<anonymous> (/bitbucket/arduino_server/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:574:13)
    at emitOne (events.js:77:13)
    at SerialPort.emit (events.js:169:7)
    at SerialPort.module.exports.raw (/bitbucket/arduino_server/node_modules/johnny-five/node_modules/serialport/lib/parsers.js:7:13)

rwaldron, 2016-10-11:
Hi! I can't be useful to you without seeing the code that triggered this error :)

rwaldron, 2016-12-08:
Please follow up if you'd like to reopen this issue—thanks!

closed 2016-12-08 by rwaldron

greenbridge, 2016-12-25:
Hi,  

[webtest.zip](https://github.com/rwaldron/johnny-five/files/672005/webtest.zip)
This also happened to me when running a node on a Raspberry Pi 3 - Pixel. (Node.js v 4.7.0):

pi@raspberrypi:~/nodejs/webtest $ node webtest.js
   info  - socket.io started
   warn  - error raised: Error: listen EACCES 0.0.0.0:80
1482626128380 Device(s) /dev/ttyACM0
/home/pi/nodejs/webtest/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:898
  this.transport.write(new Buffer([START_SYSEX, CAPABILITY_QUERY, END_SYSEX]));
                 ^

TypeError: this.transport.write is not a function
    at Board.queryCapabilities (/home/pi/nodejs/webtest/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:898:18)
    at Board.<anonymous> (/home/pi/nodejs/webtest/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:692:14)
    at Board.g (events.js:260:16)
    at emitNone (events.js:67:13)
    at Board.emit (events.js:166:7)
    at SYSEX_RESPONSE.(anonymous function) (/home/pi/nodejs/webtest/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:186:9)
    at Board.<anonymous> (/home/pi/nodejs/webtest/node_modules/johnny-five/node_modules/firmata/lib/firmata.js:574:13)
    at emitOne (events.js:77:13)
    at SerialPort.emit (events.js:169:7)
    at SerialPort.module.exports.raw (/home/pi/nodejs/webtest/node_modules/serialport/lib/parsers.js:7:13)

AlexEul, 2017-03-16:
I discovered the reason for the issue; here is the solution:

step 1: go to the application folder, and open the given package.json file.

step 2: under the dependencies/ bundledDependenices/ developerDepende
ncies, delete johnny-five from all fields within the package.json file.

step 3: run on the command line "npm install --save johnny-five", which will download the latest version of the Johnny-Five package, and the updates the johnny-five dependency to the package.json file.

step 4: run the "entry point" file of the application