rwaldron/johnny-five #1237: Active Pullup resistor for Pin

tibysko, 2016-10-10:
Hi

First, thanks for a great lib!

I have a question regarding how to enable PullUp resistor for the Pin component. According to the Arduino documentation I should be able to enable the pullUp by doing: 

```
pinMode(pin, INPUT);           // set pin to input
digitalWrite(pin, HIGH);       // turn on pullup resistors
```

I have used the same aproach as the Button component, but it does not seem to work.
```
.....
// Enable the pullup resistor after setting pin mode
this.io.digitalWrite(13, this.io.HIGH);
var pin = new five.Pin(13);

pin.read(function(error, value) {
  console.log(value); // This should write 1
});
```

Sadly I can't use the Button component since i need to read the value of the Pin. 

What am I doing wrong, any thoughts? 

Regards Ola

soundanalogous, 2016-10-11:
Are you using and Arduino Uno or other board with the onboard LED wired to pin 13. If so you can't enable a pull-up on that pin.

dtex, 2018-01-05:
There was no response so I'm assuming @soundanalogous identified the issue. @tibysko feel free to re-open if necessary.

closed 2018-01-05 by dtex