rwaldron/johnny-five #1242: Support for pull-down resistors on platforms that have them

andrewn, 2016-10-23:
On platforms such as Raspberry Pi, [some pins have both pull-up and pull-down resistors](http://wiringpi.com/pins/special-pin-functions/) but currently there's no way to enable pull-down resistors using Johnny-Five.

Enabling a pull-up resistor works by writing `HIGH` to the pin in `INPUT` mode. I'm proposing that writing `LOW` should enable the built-in pull-down resistor.

I've created an [issue](https://github.com/nebrius/raspi-io/issues/71) and [pull request](https://github.com/nebrius/raspi-io/issues/72) to support this behaviour in [Raspi-IO](https://github.com/nebrius/raspi-io).

I'm going to open a pull request with a patch enabling this but I think an issue is useful for discussing if this is the right approach and something you'd support.

revolunet, 2017-03-18:
looks like its been merged 👍

fivdi, 2017-04-17:
Now that #1243 has been merged I think this issue can be closed.

rwaldron, 2017-04-17:
@fivdi thanks for the follow up

closed 2017-04-17 by rwaldron