rwaldron/johnny-five #1245: Ping HC-SR04 timeout error

gpaixao, 2016-10-24:
I'm work with HC-SR04 and my connections is:

Board:GND -> Sensor:GND
Board:˜10(PWM) -> Sensor:Trig
Board:˜9(PWM) -> Sensor:Echo
Board:5V -> Sensor:Vcc

 I searched for instructions but found nothing plausible. Anyone, can you help me? pls 😄 

### My log error
```
daybot node server.js                                                                                                                                              node6.4.0  go1.6  ruby-2.3.0
1477330214600 Connected /dev/cu.usbmodem1421  
1477330224607 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:163:17)
    at Board.log (/Users/guilhermepaixao/Projects/src/github.com/gpaixao/daybot/node_modules/johnny-five/lib/board.js:630:8)
    at Board.(anonymous function) [as error] (/Users/guilhermepaixao/Projects/src/github.com/gpaixao/daybot/node_modules/johnny-five/lib/board.js:641:14)
    at Board.<anonymous> (/Users/guilhermepaixao/Projects/src/github.com/gpaixao/daybot/node_modules/johnny-five/lib/board.js:390:14)
    at tryOnTimeout (timers.js:232:11)
    at Timer.listOnTimeout (timers.js:202:5)
```

rwaldron, 2016-10-24:
The error has nothing specifically to do with HCSR04. Here's a few things to do and try: 

1. If you're using an Arduino, you absolutely **must** have `PingFirmata` uploaded to the board. See [johnny-five.io/api/proximity/#pingfirmata](http://johnny-five.io/api/proximity/#pingfirmata) for more information there. 
2. Make sure your wiring matches this example: http://johnny-five.io/examples/proximity-hcsr04/

gpaixao, 2016-10-24:
tks

gpaixao, 2016-11-18:
Sorry but I lost my Arduino :/

rwaldron, 2016-11-18:
Oh no! That's awful :( 

I'm going to close this for now, we can reopen it in the future if you ever need to.

closed 2016-11-18 by rwaldron