rwaldron/johnny-five #1246: pin.read() always returns 1.

derencel20, 2016-10-31:
[image]

I connected my pin #11 to a DUAL QUAD-INPUT NAND GATE (in it's output Y1), specifically HD74SLS20P (see DIAGRAM for Y1). I think that when the NAND gate's output becomes false, it should return 0, but it doesn't.

derencel20, 2016-10-31:
[image]
When Y1 returns 0, shouldn't digital read also return 0?

rwaldron, 2016-10-31:
> When Y1 returns 0, shouldn't digital read also return 0?

Yes, absolutely—but it's not clear that all conditions are being met. What are 1A, 1B, 1C and 1D connected to? They must all be "on" for a 1Y to have a value of 0. Otherwise the result is always 1:

[image]

Side-note: Google produces no results when searching for "HD74SLS20P", can you link to it?

derencel20, 2016-10-31:
it should be all lower case for google to have results.. hahaha. hd74ls20p

derencel20, 2016-10-31:
Here's the link.
http://pdf1.alldatasheet.com/datasheet-pdf/view/247370/RENESAS/HD74LS20P.html

derencel20, 2016-10-31:
Here's the schematic:

[image]

What i want to do is read the output of the nand gate.

Here's the actual connection. (The NAND gate in the schematic is on RIGHT breadboard and the inputs on the LEFT of the NAND gate)

[image]

derencel20, 2016-10-31:
The NAND gates purpose is that it'll make the LOAD's state 0. So that the 74190N will reset counting to it's preset value.

rwaldron, 2016-10-31:
Do you have a multimeter or logic analyzer? If so, can you analyze the 1Y pin and confirm that it's in the expected state when the code produces wrong values?

derencel20, 2016-10-31:
I can confirm. Y1 is connect connected both to the pin #11 and the LOAD inputs of the two counters

rwaldron, 2016-10-31:
Sorry, that's not what I was asking. Do you have the means to measure the actual output voltage of Y1?

derencel20, 2016-10-31:
No. T_T

rwaldron, 2016-10-31:
I'm trying to understand the output voltage for that chip. The datasheet appears to have very low voltage outputs. Also, this chip appears to be EOL https://www.renesas.com/en-us/products/general-purpose-logic/hd74ls/device/HD74LS20P.html

rwaldron, 2016-10-31:
From arduino docs: 

>the Arduino (Atmega) will report HIGH if:
> - a voltage greater than 3 volts is present at the pin (5V boards);
>
> the Arduino (Atmega) will report LOW if:
> - a voltage less than 3 volts is present at the pin (5V boards);

Maybe I'm misunderstanding the datasheet...?

derencel20, 2016-11-01:
Wait sir. I'm getting the output of the counter which is sn74ls190n.

derencel20, 2016-11-01:
maybe, the voltage is always higher than 3V?

derencel20, 2016-11-01:
I am currently using this.

derencel20, 2016-11-01:
What IC could you suggest sir? for dual quad-input NAND gate IC?

derencel20, 2016-11-01:
Is this related to the problem? It's stating something like `decoupling capacitors`.

derencel20, 2016-11-01:
http://www.doctronics.co.uk/4012.htm

derencel20, 2016-11-03:
My teacher said it has something to do with propagation delay. How can I handle this?

rwaldron, 2016-11-03:
I've ordered a pack of dual 4-input Positive NAND gates to reproduce your circuit.

rwaldron, 2016-11-16:
Sorry for the delay getting back to you. I received only one 7420 (I thought I had ordered a pack of 5 0_o) but it's enough to work with. As I begin to build a reproduction of your issue, the first thing I've noticed is the timing in the call to `this.loop`: 

```js
  this.loop(500, () => {
    setTimeout(() => {
      clk.high();
    }, 500); 
    setTimeout(() => {
      clk.low();
    }, 500);
  });
```

I'd like some clarification about this, but if I had to guess, I'd say that the intention is to create a 1Hz clock (500ms High, 500ms Low)? Whether or not that's the intention, the code above tick is executing the `clk.high()` and `clk.low()` calls in the same millisecond. To illustrate, I changed the code to: 

```js
  let count = 0;
  this.loop(500, () => {
    count++;
    setTimeout(() => {
      console.log(`${count} high ${Date.now()}`);
      clk.high();
    }, 500);
    setTimeout(() => {
      console.log(`${count} low ${Date.now()}`);
      clk.low();
    }, 500);
  });
```

And here is the result: 

```
$ node eg/pin-read-1246.js
1479308837162 Device(s) /dev/cu.usbmodem1411
1479308837171 Connected /dev/cu.usbmodem1411
1479308838758 Repl Initialized
>> 1 high 1479308839766
1 low 1479308839766
2 high 1479308840266
2 low 1479308840266
3 high 1479308840767
3 low 1479308840767
4 high 1479308841268
4 low 1479308841268
5 high 1479308841768
5 low 1479308841768
6 high 1479308842269
6 low 1479308842269
```

_(Note that there is a drift of approximately 1 millisecond, [here is further explanation](https://github.com/nodejs/node/blob/master/doc/topics/event-loop-timers-and-nexttick.md))_

The results above show that the calls are happening one right after the other in the same millisecond, but the `clk.high()` instruction is applied for less than that millisecond, because the `clk.low()` instruction immediately changes the state. 

My next steps: 

1. Reproduce with your original code
2. Determine a path forward the produces the results you're expecting.

rwaldron, 2016-11-16:
> 1. Reproduce with your original code

I quickly determined that I wouldn't be able to reproduce the entire circuit without the other ICs, but I could easily write a simulator. The following program writes the appropriate values for each state in the truth table and verifies that the expected output is received. 

```js
"use strict";
const five = require("johnny-five");
const board = new five.Board();

board.on("ready", function() {
  const a = new five.Pin(2);
  const b = new five.Pin(3);
  const c = new five.Pin(4);
  const d = new five.Pin(5);
  const output = new five.Pin(11);

  const tests = [
    //         DCBA
    { input: 0b0000, output: 1 },
    { input: 0b0001, output: 1 },
    { input: 0b0010, output: 1 },
    { input: 0b0011, output: 1 },
    { input: 0b0100, output: 1 },
    { input: 0b0101, output: 1 },
    { input: 0b0110, output: 1 },
    { input: 0b0111, output: 1 },
    { input: 0b1000, output: 1 },
    { input: 0b1001, output: 1 },
    { input: 0b1010, output: 1 },
    { input: 0b1011, output: 1 },
    { input: 0b1100, output: 1 },
    { input: 0b1101, output: 1 },
    { input: 0b1110, output: 1 },
    { input: 0b1111, output: 0 },
  ];

  let isAwaitingRead = false;
  let index = -1;
  let expecting = -1;

  this.loop(500, () => {
    ++index;

    if (index === tests.length) {
      index = 0;
    }
    isAwaitingRead = true;

    let test = tests[index];

    a.write(test.input & 1);
    b.write((test.input >> 1) & 1);
    c.write((test.input >> 2) & 1);
    d.write((test.input >> 3) & 1);

    expecting = test.output;
  });

  output.read((error, value) => {
    if (isAwaitingRead && expecting === value) {
      console.log(`${index}, ${tests[index].output}, ${value}`, tests[index].output === value ? "PASS" : "FAIL");
      isAwaitingRead = false;
    }
  });
});
```

[image]

Here is the resulting output from a test run: https://gist.github.com/rwaldron/116c21231102707647e87670ef271d6d

I believe the timing in the loop function may be the cause of your unexpected output from 1Y.

derencel20, 2016-11-16:
I had a chat with my teacher regarding this, and he said, I used too much ICs. hahaha. That may be the reason as to how the reading detects very low voltage..

derencel20, 2016-11-16:
I used two SN74LS47N, one HD74LS190N, one SN74LS190N, one HD74LS20P, and one HD74LS04P.

With this configuration:
[image]

derencel20, 2016-11-16:
And additional two COMMON ANODE - SEVEN SEGMENT LED DISPLAY, and 3 led lights.

derencel20, 2016-12-04:
@rwaldron Is it required to upload the code everytime I use it? Because if I use the default arduino IDE to upload the C syntax code, I only have to upload it once.

I followed the instructions here: https://www.youtube.com/watch?v=8s2--hfsJDY&t=653s

derencel20, 2016-12-04:
Whereas, If I upload the firmata, I have to upload the code everytime I use it.

rwaldron, 2016-12-05:
> Whereas, If I upload the firmata, I have to upload the code everytime I use it.

I don't know what you mean here? Uploading StandardFirmata to the Arduino only happens once. When using Johnny-Five with an Arduino, your JavaScript program runs on your local computer and communicates via a serial port. Alternatively, you can use a Tessel 2, Raspberry Pi, BeagleBone Black, Intel Edison or Joule, if you want to run the code on the board itself.

derencel20, 2016-12-06:
Yeah, I want to store the code in the board itself. I observe that once I restart the board, it flashes all the info. Is this behavior normal ?

rwaldron, 2016-12-06:
> Yeah, I want to store the code in the board itself.

The JavaScript code? That's not possible when using Johnny-Five with an Arduino. As I mentioned above, you'd need to use a board that runs Linux and Node.js, ie: Tessel 2, Raspberry Pi, BeagleBone Black, Intel Edison or Joule, Chip, 

> I observe that once I restart the board, it flashes all the info.

I don't really follow—do you mean everytime you run your JS code locally? If so, then that's not flashing the board, it's making a serial connection

dtex, 2018-01-16:
@derencel20 I'm going to close this since we haven't heard back. Feel free to re-open if necessary.

closed 2018-01-16 by dtex