rwaldron/johnny-five #1247: Using speed/acceleration capable boards.

svbatalov, 2016-11-01:
I am playing with J5 + [Pololu Maestro](https://www.pololu.com/product/1354) servo controller. This board supports setting servo speed/acceleration via dedicated commands. Is there a way to reimplement `Servo(s).to()` to use these commands instead of Animation? This would save some CPU cycles and result in smoother movement (since Standard servo controller in J5 is restricted to integer angles).

I am using [node-maestro-board-io](https://github.com/svbatalov/node-maestro-board-io) IO plugin.

rwaldron, 2016-11-03:
Yes, all that needs to be done is a `controller` definition that defines its own methods to override the defaults. I started working on a controller for the maestro, but got busy elsewhere and put it down unfinished. Let's cc @dtex for more guidance

dtex, 2016-11-03:
That's correct. So you'll be using Serial to control the servos? Cool.

There are a few commands available on the Maestro that have no equivalent in servo so you'll have to either add those or do without. Acceleration (easing) will also be different.

svbatalov, 2016-11-04:
Thanks for replies!

Unfortunately I do not see an easy way to pass my own controller implementation for use by `Servo` class and even if it is possible the `update(degrees)` method knows nothing about speed/acceleration.
So it looks like the simplest way to integrate these capabilities and keep interface consistent is to monkey patch `Servo.prototype.to()` method and support both its usesecases. Am I right?

dtex, 2016-11-04:
You can pass the controller as a javascript object. For example:

```js
var servo = new five.Servo({
  pin: 10,
  center: true,
  controller: {
    initialize: {
      value: function(opts) {
        // your initialization code here
    },
    to: {
      writable: true,
      value: function(degrees, time, rate) {
        // your implementation of to
      }
    }
  }
});
```

But what would be even better is if you add your controller to a fork of Johnny-Five and submit a pull request. That way everyone can benefit \o/

svbatalov, 2016-11-07:
Hmm.. This does not work for `Servo`. Here are relevant lines from `lib/servo.js` (branch master):
```js
function Servo(opts) {
  // ... skipped ...
  /**
   * Used for adding special controllers (i.e. PCA9685)
   **/
  controller = typeof opts.controller === "string" ?
    Controllers[opts.controller] : Controllers.Standard;

  Object.defineProperties(this, Object.assign({}, controller, {
     // ... getters for history, last and position
  }));

  this.initialize(opts);
  // etc ...
}
```
So `controller` defaults to `Standard` if I pass object. Maybe I'm looking at the wrong branch?

dtex, 2016-11-07:
Oh snap. I remembered that part differently. I'm sorry I sent you down the wrong path.

rwaldron, 2016-11-07:
> So controller defaults to Standard if I pass object.

Definitely a bug! This just needs to be updated to the new way. I can fix that asap and unblock you.

rwaldron, 2016-11-07:
@svbatalov can you please npm install j5 v0.10.5? Thanks!

svbatalov, 2016-11-07:
Ok, now it works. Thank you both for your assistance!
Btw, my fork was 1000's one! :tada:

closed 2016-11-07 by svbatalov