rwaldron/johnny-five #1248: npm not available on Galileo gen 1?

grv2k8, 2016-11-09:
Hi,

I can't find or run npm from my SSH session to my board. All the tutorials I've read have instructions to run this line:
`npm install johnny-five galileo-io`

But I'm getting errors such as "npm: command not found". This is running the latest yocto image and has node installed (v4.4.3). Any help would be appreciated.

rwaldron, 2016-11-09:
That's strange, can you link me to the yocto image that you're referring to? Thanks

rwaldron, 2016-11-09:
Also, ssh to the board and try this: `curl https://www.npmjs.org/install.sh | sh`

grv2k8, 2016-11-14:
Hi Rick,

Pretty sure this is where I got the OS image from:
https://software.intel.com/en-us/get-started-galileo-windows-step1

And trying that SSH command gives me this error:
[image]

I tried pasting the contents of install.sh into a local shell file and ran it, and this is what I'm seeing:
[image]

rwaldron, 2016-11-14:
Oh damn, I forgot that's the BusyBox image. Galileo-IO and Johnny-Five don't support that image anymore, because Intel themselves stopped supporting it.

closed 2018-08-28 by stale[bot]