rwaldron/johnny-five #1249: can't get mpr121 to work

oallouch, 2016-11-14:
Hi,

I can't get an Adafruit mpr121 shield to work. It works in standard c++ Arduino using the Adafruit library.
I uploaded the lastest firmata v2.5.4 .
What can I do to diagnosis this issue. Is there a verbose log or something like this ?
Here's my simple init:

			keypad = new five.Keypad({
				address: 0x5D,
				controller: "MPR121_SHIELD"
			});
Thx a lot in advance,

Olivier

rwaldron, 2016-11-14:
Unfortunately, I'm away from home so I can't assemble this and test today. I just ordered one of these for delivery here tomorrow and can help get you unblocked when it arrives. I appreciate your patience

oallouch, 2016-11-14:
Are you kidding !?! :) It's me who appreciate what you do !
You've already been great in the past for me (with a keypad for qTouch)
In the meantime, I'll try some chrome serial api (I'm in nw.js)
Thx a lot in advance.

rwaldron, 2016-11-14:
I really appreciate that

rwaldron, 2016-11-15:
Ok, so despite paying for next day delivery, Amazon order status shows that it will arrive tomorrow.

rwaldron, 2016-11-16:
The MPR121 shield arrived and I just assembled it. Clarifying question: Why `address: 0x5D,`? When I looked at Adafruit's guide, they have: 

> - ADDR not connected: 0x5A
> - ADDR tied to 3V: 0x5B
> - ADDR tied to SDA: 0x5C
> - ADDR tied to SCL: 0x5D

Is your `ADDR` pin tied/connected to the `SCL` pin?

While I wait to hear back, I'm going to go ahead and try both default address and `0x5D` (with expected wiring)

rwaldron, 2016-11-16:
With the following program, I tapped each contact on the left-most circle, starting from the top left, moving left to right, line by line, with shield held like this:

[image]

```js
const five = require("johnny-five");
const board = new five.Board();

board.on("ready", () => {
  const touchpad = new five.Touchpad({
    controller: "MPR121_SHIELD",
    sensitivity: {
      press: 0.5,
      release: 0.9,
    },
  });

  touchpad.on("press", (event) => {
    console.log(event.which);
  });
});
```

Result: 

```
$ node eg/keypad-MPR121_SHIELD-1249.js
1479315572728 Device(s) /dev/cu.usbmodem1411
1479315572736 Connected /dev/cu.usbmodem1411
1479315574449 Repl Initialized
>> [ 11 ]
[ 10 ]
[ 9 ]
[ 8 ]
[ 7 ]
[ 6 ]
[ 5 ]
[ 4 ]
[ 3 ]
[ 2 ]
[ 1 ]
[ 0 ]
>>
```

Note that I adjusted the sensitivity: the defaults seemed to be overly sensitive and with false positives for other pads (which is possible when pressing directly on the board)

oallouch, 2016-11-16:
That's weird.
I'll try again tomorrow.
The address is 0x5d. I need more than 12 pads.
Maybe it's because I'm in nw.js . Note that the qTouch (also I2C) works fine.
Anyway, I have a plan B for this project. Chrome Serial API is fine.
But for the next one: back to johnny-five !

Thx again anway

rwaldron, 2016-11-16:
> Maybe it's because I'm in nw.js 

This should definitely work :|

You made sure to do `npm rebuild`, right? (Sorry if that's an obvious question). I'm going to try to whip up an nw.js thing and see if I can help more.

rwaldron, 2016-11-16:
Sorry, I meant `nw-gyp`!! https://github.com/nwjs/nw.js/wiki/using-node-modules#3rd-party-modules-with-cc-addons

rwaldron, 2016-11-16:
Specifically, I had to do: 

```
cd node_modules/johnny-five/node_modules/serialport/;
nw-gyp rebuild --target=0.18.7;
```

(I installed nw.js v 0.18.7)

oallouch, 2016-11-16:
in the latest nw.js version, you don't have to do that anymore (I think)
The Arduino already is connected.

rwaldron, 2016-11-16:
Oh, I didn't see that anywhere and just assumed that I still had to do that (as I had last time I did this). I just made this video https://youtu.be/UK8OKsd26YM

rwaldron, 2016-11-16:
index.html:

```html
<!DOCTYPE html>
<html>
  <head>
    <title>j5-nw</title>
  </head>
  <body>
    <h1 id="key"></h1>
    <script src="index.js"></script>
  </body>
</html>
```

package.json: 

```json
{
  "name": "j5-nw",
  "main": "index.html"
}
```

index.js: 

```js
"use strict";
const five = require("johnny-five");
const board = new five.Board({
  debug: false,
  repl: false,
});
const win = nw.Window.get();

win.showDevTools();

console.log(document);

board.on("ready", () => {
  const touchpad = new five.Touchpad({
    controller: "MPR121_SHIELD",
    sensitivity: {
      press: 0.5,
      release: 0.9,
    },
  });

  touchpad.on("press", (event) => {
    document.getElementById("key").innerHTML = event.which[0];
  });
});

// ~/Downloads/nwjs-sdk-v0.18.7-osx-x64/nwjs.app/Contents/MacOS/nwjs .
```

rwaldron, 2016-11-16:
I found the announcement mentioning the nw-gyp thing—thanks for the heads up

oallouch, 2016-11-16:
How can you rock more ?!!
I'll try that.
Btw, you don't use nwjs-j5-fix ?

rwaldron, 2016-11-16:
> nwjs-j5-fix

I totally forgot!

rwaldron, 2016-11-17:
Update: I wired up two of these to test it all out and everything is working correctly, with only one minor hiccup along the way: I accidentally wired the `addr` for the second shield incorrectly and as a result I got no output from any shields and it took me some time to figure that out—whoops.

Here's the correct wiring: 

[image]

index.html: 

```html
<!DOCTYPE html>
<html>
<head>
  <title>j5-nw</title>
  <style type="text/css">
    * {
      box-sizing: border-box;
      -webkit-touch-callout: none;
      -webkit-text-size-adjust: none;
      -webkit-tap-highlight-color: rgba(0,0,0,0);
      -webkit-user-select: none;
    }

    body, html {
      margin: 0px;
      height: 100%;
      position: fixed;
    }

    body {
      overflow: auto;
    }

    .touchpad {
      display: inline-block;
      border: 1px solid black;
      width: 270px;
      height: 220px;
    }
    .pad {
      /*border: 1px solid black;*/
      background-color: lightblue;
      display: inline-block;
      float: left;
      width: 85px;
      height: 50px;
      line-height: 50px;
      margin-left: 3px;
      margin-top: 3px;
      text-align: center;
    }

    .down {
      background-color: pink;
    }
  </style>
</head>
<body>
  <div id="a" class="touchpad">
    <div class="pad" id="a11">11</div>
    <div class="pad" id="a10">10</div>
    <div class="pad" id="a9">9</div>
    <div class="pad" id="a8">8</div>
    <div class="pad" id="a7">7</div>
    <div class="pad" id="a6">6</div>
    <div class="pad" id="a5">5</div>
    <div class="pad" id="a4">4</div>
    <div class="pad" id="a3">3</div>
    <div class="pad" id="a2">2</div>
    <div class="pad" id="a1">1</div>
    <div class="pad" id="a0">0</div>
  </div>
  <div id="b" class="touchpad">
    <div class="pad" id="b11">11</div>
    <div class="pad" id="b10">10</div>
    <div class="pad" id="b9">9</div>
    <div class="pad" id="b8">8</div>
    <div class="pad" id="b7">7</div>
    <div class="pad" id="b6">6</div>
    <div class="pad" id="b5">5</div>
    <div class="pad" id="b4">4</div>
    <div class="pad" id="b3">3</div>
    <div class="pad" id="b2">2</div>
    <div class="pad" id="b1">1</div>
    <div class="pad" id="b0">0</div>
  </div>
  <script src="index.js"></script>
</body>
</html>
```

index.js: 

```js
"use strict";
const five = require("johnny-five");
const board = new five.Board({
  debug: false,
  repl: false,
});
const win = nw.Window.get();

win.showDevTools();
win.moveTo(0, 0);
win.resizeTo((270 * 2 + 10), 250);

board.on("ready", () => {
  win.focus();

  const pads = {
    a: Array.from(document.querySelectorAll("#a .pad")).reverse(),
    b: Array.from(document.querySelectorAll("#b .pad")).reverse(),
  };

  const controller = "MPR121_SHIELD";
  const sensitivity = {
    press: 0.5,
    release: 0.9,
  };
  const a = new five.Touchpad({
    address: 0x5A,
    controller,
    sensitivity
  });

  const b = new five.Touchpad({
    address: 0x5B,
    controller,
    sensitivity
  });

  a.on("change", (event) => {
    toggle({
      pads: pads.a,
      event,
    });
  });

  b.on("change", (event) => {
    toggle({
      pads: pads.b,
      event,
    });
  });

  function toggle(target) {
    target.pads.forEach(pad => pad.classList.remove("down"));
    if (target.event.type === "down") {
      target.pads[target.event.which[0]].classList.add("down");
    }
  }
});
// ~/Downloads/nwjs-sdk-v0.18.7-osx-x64/nwjs.app/Contents/MacOS/nwjs .
```

Here's a video of the multiple shield demo: 

https://www.youtube.com/watch?v=Hiy1Ru705ug

oallouch, 2016-11-17:
This looks great.
You should definitely put this in the docs.
For my problem, I must have done something wrong.
The symptoms were the same when I tried with another MPR121 (the seeedstudio's one), almost 6 months ago.
I'll work something out with the Chrome Serial API.

Anyway, thx A LOT (again), for what you did for me...and for the makers community.

Olivier Allouch
Paris

dmrieder, 2018-10-20:
Thanks for everything you posted here. If you have a moment, can you help me revise your index.js so that it writes to the node console? I have 4 mpr121s connected, and I'm not sure how to revise your index.js for the console.

dtex, 2018-11-29:
This looks like it was resolved quite some time ago so I'm going to close it.

closed 2018-11-29 by dtex