rwaldron/johnny-five #1250: [Question] Is TI Launchpad really supported

lpan, 2016-11-18:
It is mentioned briefly in [README](https://github.com/rwaldron/johnny-five/blame/master/README.md#L68), but I can't find it on the [platform support page](http://johnny-five.io/platform-support/)

rwaldron, 2016-11-18:
That's a damn good question. 

Here's what I've done and learned this morning: 

- Installed [Energia](http://energia.nu/download/)
- Installed [the drivers](https://github.com/energia/Energia/wiki/Getting-Started)
- I downloaded [Energia repo](https://github.com/energia/Energia), and copied the `libraries/` dir to `~/Downloads/Energia/` (replace or merge)

...While trying to understand the pin naming, I went digging through code to find the Boards.h file and found this: 

```
// MSP430G2553 Value Line
#elif defined(__MSP430G2553__)
// Baudrate only up to 9600, currently does NOT support Firmata
#define TOTAL_ANALOG_PINS       8
#define TOTAL_PINS              20 
#define VERSION_BLINK_PIN       2
#define IS_PIN_DIGITAL(p)       ((p) >= 3 && (p) < TOTAL_PINS && (p)!=16)
#define IS_PIN_ANALOG(p)        ((p) >= 3 && (p) < 8)
#define IS_PIN_PWM(p)           digitalPinHasPWM(p)
#define IS_PIN_SERVO(p)         IS_PIN_PWM(p)
#define IS_PIN_I2C(p)           ((p) == 14 || (p) == 15)
#define PIN_TO_DIGITAL(p)       (p)
#define PIN_TO_ANALOG(p)        ((p))
#define PIN_TO_PWM(p)           PIN_TO_DIGITAL(p)
#define PIN_TO_SERVO(p)         (p)
//#define TOTAL_PORTS				2
```

I presently have the MSP430G2553, so this jumps out: 

> `// Baudrate only up to 9600, currently does NOT support Firmata`

rwaldron, 2016-11-18:
I've updated the baudrate everywhere it needs to be updated but no data flows from the serialport once it's opened.

rwaldron, 2016-11-18:
According to the Boards.h file, these are supported: 

[image]

rwaldron, 2016-11-18:
I just ordered both

soundanalogous, 2016-11-19:
I take it Energia ships with it's own version of Firmata as those boards are not defined in the Boards.h file that ships with Firmata.

lpan, 2016-11-19:
@rwaldron thank you so much for the fast reply!

closed 2018-08-23 by lpan