rwaldron/johnny-five #1252: More than one "Multi" I2C device connected in the same I2C bus

aabm00, 2016-11-23:
Hi 

The SHT31-D temperature and humidity sensor has two I2C addresses. Can I connect both of them in the same I2c bus in an Arduino Yun?

There are some example where shows how to connect more than one "Multi" device (same type or differents) in the same I2C bus with J5 directly in an Arduino board

Thanks

soundanalogous, 2016-11-23:
As long as each I2C device has a unique address, you should be able to connect up to 16 devices to a single I2C port on a board. Of course bandwidth may be an issue as the number of devices increases, but 2 should be no problem. You will need to add a couple of pull-up resistors unless they are already included on each I2C module you are attaching. See the second answer [here for a wiring diagram](http://electronics.stackexchange.com/questions/25278/how-to-connect-multiple-i2c-interface-devices-into-a-single-pin-a4-sda-and-a5).

rwaldron, 2016-11-23:
@soundanalogous thanks! 

To illustrate: 

```js
var a = new five.Multi({ controller: "SHT31D", address: 0x44 });
var b = new five.Multi({ controller: "SHT31D", address: 0x45 });
```

For the second to work, you'll need to connect the ADDR pin to a HIGH voltage

closed 2016-11-23 by rwaldron

aabm00, 2016-11-24:
Thanks you both @soundanalogous and @rwaldron for the answers.

Sorry, I know that the post is closed but I have found another problem related with. 
As I mentioned I need four PT-100 amplifier but the only I have found was the MAX31865 https://www.adafruit.com/products/3328 and it uses only the SPI protocol.
Looking for information about how to implement it in J5 seems that Firmata doesn't still support the SPI protocol, so some questions came up.

1- ¿There are some way to do it, if finally I can't find an I2C PT-100 amplifier?
2- ¿Can I have UART, SPI and I2C devices connected to the same arduino board (In my case YUN) working together?

Thanks again

soundanalogous, 2016-11-24:
You can have UART, SPI and I2C all connected simultaneously. Firmata won't have SPI support until I get some help on the Firmata SPI proposal: https://github.com/firmata/protocol/pull/27. I really need an expert in SPI to weigh in. It's not as easy to support as I2C because it's not as standardized as I2C and thus there are several edge cases making it very difficult to support in the generic way that Firmata needs.

aabm00, 2016-11-25:
Thanks !!!

I hope that soon someone will help you with the implementation. So the range of devices that can be used is broader.

Good luck