rwaldron/johnny-five #1254: using johnny-five with node-red causes reboots on raspberry pi.

DannyBoyBroadSword, 2016-11-29:
```
sudo node-red-start

Start Node-RED
 
Once Node-RED has started, point a browser at http://192.168.1.184:1880
On Pi Node-RED works better with the Firefox browser
 
Use   node-red-stop                          to stop Node-RED
Use   node-red-start                         to start Node-RED again
Use   node-red-log                           to view the recent log output
Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot
 
To find more nodes and example flows - go to http://flows.nodered.org
 
Started Node-RED graphical event wiring tool..
Welcome to Node-RED
===================
29 Nov 23:40:28 - [info] Node-RED version: v0.15.2
29 Nov 23:40:28 - [info] Node.js  version: v4.6.2
29 Nov 23:40:28 - [info] Linux 4.4.24-v7+ arm LE
29 Nov 23:40:29 - [info] Loading palette nodes
root : TTY=unknown ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/python -u /usr/lib/node_modules/node-red/nodes/core/hardware/nrgpio.py info
pam_unix(sudo:session): session opened for user root by (uid=0)
pam_unix(sudo:session): session closed for user root
29 Nov 23:40:37 - [info] Settings file  : /root/.node-red/settings.js
29 Nov 23:40:37 - [info] User directory : /root/.node-red
29 Nov 23:40:37 - [info] Flows file     : /root/.node-red/flows_raspberrypi.json
29 Nov 23:40:37 - [info] Server now running at http://127.0.0.1:1880/
29 Nov 23:40:37 - [debug] red/runtime/nodes/credentials.load : no user key present
29 Nov 23:40:37 - [debug] red/runtime/nodes/credentials.load : using default key
29 Nov 23:40:37 - [info] Starting flows
29 Nov 23:40:37 - [info] Started flows
29 Nov 23:40:50 - [info]  Enabled node types:
29 Nov 23:40:50 - [info]  - gpio in
29 Nov 23:40:50 - [info]  - gpio out
29 Nov 23:40:50 - [info]  - nodebot
29 Nov 23:40:50 - [info]  - johnny5
29 Nov 23:40:56 - [error] [function:board] TypeError: Cannot read property 'Board' of undefined
29 Nov 23:42:51 - [info] Stopping flows
29 Nov 23:42:51 - [info] Stopped flows
29 Nov 23:42:51 - [info] Starting flows
1480462972554 Device(s) RaspberryPi-IO
29 Nov 23:42:52 - [info] Started flows
1480462972675 Connected RaspberryPi-IO
Connection to 192.168.1.184 closed by remote host.
Connection to 192.168.1.184 closed.
```
[image]

DannyBoyBroadSword, 2016-11-29:
As soon as I run a node-red flow with johnny-five it simply reboots my raspberry pi? Very weird? It is plenty powered, very confused on what would make this happen? Thanks.

monteslu, 2016-11-30:
@DannyBoyBroadSword which version of the pi hardware, and which version or raspi-io ?

DannyBoyBroadSword, 2016-11-30:
Raspberry Pi 2 Version B v1.1, latest raps-io

monteslu, 2016-11-30:
Got it working.  Though I'll admit not very straight forward :)

First ran as normal pi user
Then installed node-red-contrib-gpio in pi user's .node-red directory.
Then installed raspi-io in same directory.
Then copied .node-red folder to /root
Then modified `/lib/systemd/system/nodered.service` to run as root
Then `sudo systemctl daemon-reload` to pick up that service change.
Then finally ran `sudo node-red-start`

Works fine, though just for the first deploy.  

After that `Error: Error Resource temporarily unavailable Cannot lock port`  which I don't think is related to raspi-io or johnny-five.

If that helps, let's close this issue as it's likely an issue in node-red or in node-red-contrib-gpio.

DannyBoyBroadSword, 2016-11-30:
Okay. Time to forward that issue I to their stack of problems. Thanks.

monteslu, 2016-11-30:
before opening an issue on node-red see if you can get as far as I did.  Digging a little deeper I think the redeploy problem is coming from something in serialport: https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/39d01039376c29b2d317d06e59745a2dee07d69b/src/serialport_unix.cpp#L317

Serialport shouldn't be even touched at this point, but raspi-serial uses serialport it and raspi-io has a dependency on raspi-serial.  I'll try and isolate that into a simple test case tomorrow.

DannyBoyBroadSword, 2016-11-30:
Serialport is one of the most finicky of all the libs. Also when you have
different binaries for different types of raspi-io,node,node-red,node-gyp.
Things can get out of hand very quickly. Thanks ks for hopping on this.

DannyBoyBroadSword, 2016-11-30:
Locking port might come from that the port has t been closed or there are
weird permissions. This is most likely a serial-port issue now at least.

monteslu, 2016-11-30:
Locking was made optional on serialport back in June.  I think we may need the ability to pass that option down into raspi-io.  Then in turn, I can use that option in the node-red nodes.

DannyBoyBroadSword, 2016-11-30:
Okay I tried todo everything that you did and I got this 
```
30 Nov 13:37:48 - [warn] ------------------------------------------------------
30 Nov 13:37:48 - [warn] [gpio] Error: Cannot find module 'serialport'
30 Nov 13:37:48 - [warn] ------------------------------------------------------
30 Nov 13:37:48 - [info] Settings file  : /root/.node-red/settings.js
30 Nov 13:37:48 - [info] User directory : /root/.node-red
30 Nov 13:37:48 - [info] Flows file     : /root/.node-red/flows_raspberrypi.json
30 Nov 13:37:48 - [info] Creating new flow file
30 Nov 13:37:48 - [info] Starting flows
30 Nov 13:37:48 - [info] Started flows
30 Nov 13:37:48 - [info] Server now running at http://127.0.0.1:1880/
```
And furthrmore johnny-five node did not pop up on the node-red gui

monteslu, 2016-12-05:
@DannyBoyBroadSword I just bumped node-red-contrib-gpio to put serialport back in as a top level dependency.  I think we're still going to need raspi-io changes to make this reliable though.

dtex, 2018-01-05:
Hiya @monteslu I'm on a mission from God to clean up Johnny-Five issues. You mention that there might need to be some work done on rasp-io. Do you know if that happened?

closed 2019-02-04 by stale[bot]