rwaldron/johnny-five #1255: servos.to(degrees,ms) incorrect direction on inverted servo

nidersvk, 2016-12-01:
Hi,

I have a two servos setup where one of the servos is inverted. When using servos.to(degrees) function both servos moves correctly to desired position. But when I add the "ms" parameter the inverted servo moves in opposite direction as if the "invert" parameter of the inverted servo would be ignored.

```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {

    var left = new five.Servo({ pin: 3, invert: false, startAt: 80 });
    var right = new five.Servo({ pin: 5, invert: true, startAt: 80 });

    var servos = new five.Servos([left, right]);

    this.repl.inject({
        servos: servos,
        left: left,
        right: right
    });
});

```
Then in REPL console:
```
servos.to(120) // works as expected
servos.to(120,1500) // moves inverted servo in wrong direction

```
Board:
Arduino Nano

Johnny-Five Version:
0.10.6

dtex, 2016-12-01:
Thank you for the report. I'm AFK but will take a look this evening. It should be an easy fix, would you like to take a stab at it?

dtex, 2016-12-02:
Okay, so you've uncovered something more. The fix for your issue is actually pretty easy but a big change was made sometime in the past year that caused this and some other stuff to go awry. Unit tests weren't catching it because these tests are in an extended set that only run when someone explicitly does so.

nidersvk, 2016-12-02:
Yes, the [fix](https://github.com/nidersvk/johnny-five/commit/582f50de6ca4825d7d5209b95d54419e4474819d) was easy (at least for my case). Don't know if the [change](https://github.com/nidersvk/johnny-five/commit/582f50de6ca4825d7d5209b95d54419e4474819d) I've made may break other functionality.

PR #1377 (2017-08-18): Servo values were being inverted twice when Servo.to() was called with a time
Fixes #1255
commit 68cde288 (2017-08-18): Servo values were being inverted twice when to was called with a time

Fixes #1255
closed 2017-08-18 by rwaldron