rwaldron/johnny-five #1258: cleanup/destroy components?

gbaumgart, 2016-12-04:
Hi,
i understand now that there is no board disconnect method but how can we properly destruct components like buttons or sensors? I had a look at the source code, none of the components has lifecycle methods for "destroy".

For instance button sets an interval somewhere in initialize but i can't figure out a way to clean it up properly. 

Possibly I still don't understand the whole thing and I'd be delighted to know how prevent leaks. Do I have to cache just each component per component's option hash ? 

thanks a lot for any hint.
g

gbaumgart, 2016-12-04:
such problems come up if you wanna integrate johnny-five in an authoring software where users freely arrange components: 

[image]

monteslu, 2016-12-04:
@gbaumgart there is not foolproof way to do this that I've found.  I think I managed to clean up most the j5 instance, but did get a report of something still leaking:  https://github.com/monteslu/node-red-contrib-gpio/blob/master/lib/nodebotNode.js#L81-L96

Only thing that I've gotten to work 100% of the time is running j5 in its own thread/webworker and killing/restarting that thread when necessary.

gbaumgart, 2016-12-04:
@monteslu thanks a lot for the link! that brings some light. So for now I will go for component caching just.  

I am not that sure but about your cleanup method in the node-red gpio, don't you have to set the intervals to null as well (component.state.interval)? I noticed a leak in my stuff as well by not doing so.

rwaldron, 2016-12-05:
What you're asking for would require a non-trivial amount of work across several projects. I need to spend more time thinking about this before I attempt to answer your questions.

gbaumgart, 2016-12-05:
-@rwaldron : no problem, I got it done via some caching (md5(args)) for now, but yeah, some component:destroy would be awesome.

thanks for looking at it!

closed 2018-08-28 by stale[bot]