rwaldron/johnny-five #1259: USB serial on Raspberry Pi performance (?)

cylinderStudio, 2016-12-08:
I have a [robotic arm](http://www.trossenrobotics.com/robotgeek-snapper-robotic-arm) comprised of five servos. I'm running Johnny Five on a Macbook Pro and am using the Animation library to run a series of servo animations on the arm. The first segment enqueued after board.on('ready') doesn't always run perfectly, but all subsequent ones (that are triggered by button clicks in my Nodejs app) run fine. I'm connected to the Arduino running the servos via USB serial, and the servos are adequately powered by a separate power supply.

I also have Johnny Five running the identical application on a Raspberry Pi 3 and identical version of NodeJS, etc. When I connect it to the robotic arm and run my application, almost none of the animation segments run as they should. The servos act very erratically and unpredictably, often with one servo in the chain . 

Why does it run so poorly from the Raspberry Pi? Does it have to do with USB port speed? I've tried manually setting it to 115200 on the Pi application.

dtex, 2016-12-08:
On the Macbook Pro, if you wait a second or two before enqueueing the first animation segment does it play smoothly?

Re: the Raspberry Pi configuration, try reducing the fps. I suspect the Pi can't keep up with the default (but that's just a hunch).

cylinderStudio, 2016-12-08:
I'll try this when I get back to the lab tomorrow. On both the Macbook and the Pi, I did try delaying steps of the setup by a second, but I didn't fiddle with it much. I didn't think to try the fps.

dtex, 2016-12-08:
Cool. My hypothesis is that the RaspberryPi is only giving you a few frames per second or worse. That means that by the time the next render call rolls around the animation has moved on to the next cuepoint so it never reaches the target position. I've seen this with a Tessel 2 and my hexapod (18 servos) and even more glaringly when trying to control servos with animation() over an Xbee.

cylinderStudio, 2016-12-08:
That makes total sense to me, even in my limited experience.

So I delayed enqueue on the first segment by 3 seconds after the board reports ready, and I dropped the fps to 15 on all other segments. After three or four tests, that seems to have really improved things on the Macbook Pro. When the board first reports ready, the servos do an initial, somewhat strange sequence of movements, but I reckon that's to be expected. Maybe setting a start position would fix that.

As for the Pi, I delayed enqueue on the first segment by 3 seconds and dropped the fps on all segments to as low as 5, and it's still quite erratic. It improved somewhat for the first couple of segments enqueued, but it goes nuts after that. Primarily, you'll see one servo that has a bit of a load on it get behind and go into a rut where it just "ticks" a small movement every second until either it gets to it's originally targeted position, or it throws the rest of the animation(s) into chaos.

cylinderStudio, 2016-12-09:
I'm able to switch the RobotGeek Geekduino board to 3.3v, so I connected the RPi and Arduino via UART, as a possible end-around on the USB issue. No dice/same results.

dtex, 2016-12-10:
I was thinking that the FPS is limited by the processor on the Pi, not by the USB. Would you mind sharing your code? I'd like to see if I can optimize anything in Animation() to help with this.

cylinderStudio, 2016-12-10:
Oh, I didn't think of that. Gotcha.

I don't have the UI code with me, but here's the Johnny-Five code I've been tinkering with:

```
const five = require('johnny-five');
const board = new five.Board();

var base;
var shoulder;
var elbow;
var wrist;
var gripper;

var arm_animation;

board.on('ready', function() {
  base = new five.Servo({pin:3});
  shoulder = new five.Servo({pin:5});
  elbow = new five.Servo({pin:6, offset:-12});
  wrist = new five.Servo({pin:9});
  gripper = new five.Servo({pin:10});

  var joints = new five.Servos([base,shoulder,elbow,wrist,gripper]);

  arm_animation = new five.Animation(joints);

  setTimeout(function() { send_to_queue('ServiceIntent'); }, 3000);
});

// Segment definition

var extended = {
  duration: 3000,
  cuePoints: [0, 1.0],
  fps: 25,
  keyFrames: [ 
    [null, {degrees: 90,easing: "inOutQuad"}],
    [null, {degrees: 90,easing: "inOutQuad"}],
    [null, {degrees: 90,easing: "inOutQuad"}],
    [null, {degrees: 90,easing: "inOutQuad"}],
    [null, {degrees: 90,easing: "inOutQuad"}]
  ]
};

var grab_and_drop = {
	duration: 8000,
	cuePoints: [0,0.13,0.25,0.38,0.5,0.63,0.75,0.88,1.0],
	fps: 25,
	keyFrames: [
		[null,{degrees: 180,easing: "inOutQuad"},false,false,false,{degrees: 0,easing: "inOutQuad"},false,false,{degrees: 0,easing: "inOutQuad"}],
		[null,{degrees: 90,easing: "inOutQuad"},false,false,false,false,false,false,{degrees: 90,easing: "inOutQuad"}],
		[null,{degrees: 180,easing: "inOutQuad"},{degrees: 90,easing: "inOutQuad"},false,{degrees: 180,easing: "inOutQuad"},false,{degrees: 90,easing: "inOutQuad"},false,{degrees: 180,easing: "inOutQuad"}],
		[null,{degrees: 0,easing: "inOutQuad"},{degrees: 90,easing: "inOutQuad"},false,{degrees: 0,easing: "inOutQuad"},false,{degrees: 90,easing: "inOutQuad"},false,{degrees: 0,easing: "inOutQuad"}],
		[null,{degrees: 180,easing: "inOutQuad"},false,{degrees: 20,easing: "inOutQuad"},false,false,false,{degrees: 180,easing: "inOutQuad"},{degrees: 180,easing: "inOutQuad"}]
	],
	loop: true
};

// When DOM is ready, I have some jQuery buttons in a web UI trigger an enqueue of a segment with a statement like...
// arm_animation.enqueue(grab_and_drop);

```

dtex, 2016-12-10:
Your code certainly looks good. Let's find out the actual FPS...

In ```/node_modules/johnny-five/lib/animation.js``` change these lines (286-287):

````
// call render function with tweened value
this.target[Animation.render](this.tweenedValue(indices, progress));
````

to: 

````
// call render function with tweened value
this.emit("animation:render");
this.target[Animation.render](this.tweenedValue(indices, progress));
````

and in your code add a counter:

````
var frames = 0;
arm_animation.on("animation:render", function() {
  frames++;
});
````

Check ```frames``` before and after enqueueing ```grab_and_drop``` and divde by the duration. I'm guessing we're talking single digit FPS. Also, what is the frame rate if you drop the easing?

cylinderStudio, 2016-12-10:
Thanks for whipping this up. I can try it first thing Monday morning. I did try dropping the easing last night, and it had no effect, but I'll try again.

dtex, 2016-12-12:
> It's 145 both before and after enqueueing

Not sure I follow as it should be different. Perhaps you need to wait for the animation to finish before checking the value?

cylinderStudio, 2016-12-12:
Right. Tried again:

before enqueue: 143
after enqueue: 143
on segment start: 143
on segment complete: 212

dtex, 2016-12-12:
Okay, so that's about 8 fps which is low, but not so bad that it should be super wonky (maybe just jittery if your servos are fast). Also, I am assuming that the frame renderings are evenly distributed over the duration of the animation since the formula remains the same.

hmm... Maybe it is the USB after all (I'm surprised).

Let's make sure that the wonkiness is an artifact of the USB and not the frame rate. If you throttle the fps to 8 on your Macbook does it render the same as on the Pi or is it more acceptable?

Thank you for trying all this. It really helps to see how other systems work with Animation.

cylinderStudio, 2016-12-12:
No problem. Happy to help.

If I throttle to 8 fps in the segments on the Mac, the reported/computed fps drops to around 3 fps and gets more erratic.

dtex, 2016-12-12:
Oh, that's very interesting. I'm going to try replicating that tonight.

cylinderStudio, 2016-12-12:
Good deal. Here's a little more info that may or may not help. 

Those original frame rates I reported occurred when I capped the fps at 25, on both Mac and Pi. The servos seemed to be completely out of whack, even with straightforward Arduino-based tests. 

So as an experiment, I unplugged everything, reset the Geekduino, plugged everything back in, and removed the fps setting from my segments. I ran three tests on the Mac, then the Pi. 

Resulting fps ranges reported/computed:
Mac - baudrate not set/default: 24.9 to 34.7 fps
Mac - baudrate 57600: 26.5 to 29.4 fps
Mac - baudrate 115200: 21 to 34.7 fps

Pi - baudrate not set/default: 13 to 25 fps
Pi - baudrate 57600: 19 to 29 fps
Pi - baudrate 115200: 20.2 to 30 fps

The servos ran pretty well on the Mac. A few hiccups in the beginning sometimes, but they smooth themselves out.

The servos really struggle on the Pi. The ones that are running are completely out of sync, and one or two will get bogged down and just tick every second.

cylinderStudio, 2016-12-19:
As an experiment, I forked my project so that the servo animations all live in a sketch on the Arduino, and a Node app without Johnny Five that merely sends characters as commands over serial to run those animations. The animations continue to perform well when the Node app runs from my Macbook, but they perform noticeably worse when the Node app runs from the Raspberry Pi.

dtex, 2016-12-20:
If the animation code is in a custom sketch, it is unclear to me how being connected to a Pi vs your laptop could affect the performance. That custom sketch is not a variation of firmata or anything is it?

cylinderStudio, 2016-12-20:
Nope, it's a clean sketch just using the Servo library and serial.print. 

At any rate, it's pretty clear this has nothing to do with Johnny-Five, so I wanted to share. Reckon we can close this out. Thanks for your help.

closed 2016-12-20 by cylinderStudio

cylinderStudio, 2016-12-22:
It turns out not to be an issue with commands over serial at all. Whenever I just run an Arduino sketch with servo animations that's not receiving communication of any kind, it runs well when the Arduino is powered by USB from a Mac, but poorly when powered by USB from an RPi.