rwaldron/johnny-five #1260: Device or Firmware Error A timeout occurred while connecting to the Board.

wilchen558, 2016-12-09:
Hi,

I got an official Arduino Uno and I'm trying to run:
```
var five = require("johnny-five"),
    board = new five.Board();

board.on("ready", function() {
  // Create an Led on pin 13
  var led = new five.Led(13);

  // Strobe the pin on/off, defaults to 100ms phases
  led.strobe();
});
```

It works for a few seconds, then i get:

```
1481312223042 Connected /dev/cu.usbmodem1411  
1481312233046 Device or Firmware Error A timeout occurred while connecting to the Board. 

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:163:17)
    at Board.log (/www/learning/arduino/js-hello/node_modules/johnny-five/lib/board.js:630:8)
    at Board.(anonymous function) [as error] (/www/learning/arduino/js-hello/node_modules/johnny-five/lib/board.js:641:14)
    at Board.<anonymous> (/www/learning/arduino/js-hello/node_modules/johnny-five/lib/board.js:390:14)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)
```

Tried both on mac and windows with StandardFirmata and StandardFirmataPlus.

If I run:

```
var serialport = require("serialport");
serialport.list(function(err, ports) {
    console.log(ports);
});
```
I get:

```
[ { comName: '/dev/cu.Bluetooth-Incoming-Port',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/cu.usbmodem1411',
    manufacturer: 'Arduino Srl            ',
    serialNumber: '7543536303835141F040',
    pnpId: undefined,
    locationId: '0x14100000',
    vendorId: '0x2a03',
    productId: '0x0043' } ]
```

I have had no other problems with this board.

Hope you can help find a solution to this problem.

rwaldron, 2016-12-09:
Did you upload StandardFirmata to the Arduino?

soundanalogous, 2016-12-10:
@wilchen558 did you use the version of StandardFirmata that ships with the Arduino IDE or did you clone master from [firmata/arduino](https://github.com/firmata/arduino)?

wilchen558, 2016-12-11:
Yes did a clean installation of Arduino IDE and I'm using Firmata 2.5.3

rwaldron, 2016-12-12:
Any updates? I'm not sure how to help with this issue, as I can only reproduce it with a board that has something other than StandardFirmata (or similar variant)

wilchen558, 2016-12-16:
I have tried on two computers with clean installations of Arduino IDE and the latest version of Firmata, still no luck. Could it be a hardware problem?

rwaldron, 2016-12-16:
That would be my guess—as this will only happen when there is something wrong on that end.

soundanalogous, 2016-12-17:
Could also be the USB cable. Or if you connect the USB cable directly to your monitor USB port rather than computer port, or even some cheap USB hubs may cause issues.

Litz15, 2017-03-21:
I'm having the same issue, any updates on how to fix it?

dtex, 2017-03-22:
@Litz15 The only way this has been reproduced is by one of the following:
* Not having some variant firmata installed on the Arduino
* Bad USB cable
* Cheap USB hub
* Misconfigured USB drivers (Windows)

The most common of these is not having Firmata installed. Can you confirm that it is installed on the Arduino?

Litz15, 2017-03-22:
@dtex Hi! I check all those options and everything was ok. After that, I realize that when I ran the program, there was an error in the ports. It was looking for the board always in COM3, so I check the specific port for my board in arduino's IDE, in this case COM4 and put this in js code:

const board = new five.Board({ port: "COM4" });

Now everything's working fine. Thanks for your help!

dtex, 2017-03-22:
Glad to hear it's working for you now.

@wilchen558 Did you ever determine what was happening in your case?

wilchen558, 2017-03-23:
@dtex No I never got it working. Tried on Mac and PC, tried with different USB cables, tried with different versions of Firmata nothing helped.

Recently got a tip from @markvdesign

Will try this next week and report back.

> I know this is a very old thread but I had the same issue and I ended up resolving it by increasing the timeout property when initialising a new board.
>
>From this:
>const board = new five.Board({ timeout: 30 });
>
>To this:
>const board = new five.Board({ timeout: 3600 });
>
>The documentation states that timeout is in seconds but I was getting this error immediately when initialising the board. Increasing the timeout made all the difference.
>
>Note: This was using an Arduino Nano clone running on Windows 10

diegogarciaarriaza, 2017-08-07:
Hey!, do continue this post alive?
I have a similar issue. I use firmataethernet with the common port 3030.
I connect with my board, and i can keep comunication with it for severals minutes, but suddenly board is dead, no messages are shown, only stay there. If I try to force to restart the node program then i get the message: "A timeout occurred while connecting to the Board"

Maybe i lost the connection?, maybe i need to use another firmata, for example customizablefirmata?

I have installed firmataethernet on board, and I use johnny-five and etherport on the server.
Copy a pieze of my code:
```
var five = require("johnny-five");
var EtherPort = require("etherport");
var board = new five.Board({
    id: "Board_1",
    port: new EtherPort(3030),
    host: '192.168.1.101',
    timeout: 1e5
});
board.on("ready", function() {
//things
});

```
Any idea?, Thanks!!

dtex, 2017-08-07:
@diegogarciaarriaza Is it possible your board is experiencing a brown out? How are you powering it and what if anything are you powering through it (i.e. servos, motors, etc)?

diegogarciaarriaza, 2017-08-07:
I have plugged it to the wall directly, so i think always is powered correctly.
It has connected 1 motion sensor, 1 thermometer, 1 gas sensor and 1 double relay. This relay can be off all time and the problem happen too.

rwaldron, 2017-08-07:
Does the board have StandardFirmataEthernet firmware uploaded?

diegogarciaarriaza, 2017-08-07:
Yes, i think so. Board can be uploaded with firmataethernet more or less 1 month ago.
Im going to upload with firmatacustom tomorrow, and i will see if the issue persist. I read for there this second one is better now....

rwaldron, 2017-08-07:
>  I read for there this second one is better now....

I'm not sure that's true, let's ask @soundanalogous

diegogarciaarriaza, 2017-08-07:
So then?, What can i do?

soundanalogous, 2017-08-07:
It's the same code it both. Shouldn't make any difference.

What you can do is add debug statements in StandardFirmataEthernet and then keep your board connected to your computer via Serial so you can view the log output.

Uncomment [this line](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataEthernet/StandardFirmataEthernet.ino#L65) to enable serial debugging.

Then add a counter variable near the top of the file where other variables are declared:
```c++
int counter = 0;
```

Then at the end of [this if block](https://github.com/firmata/arduino/blob/master/examples/StandardFirmataEthernet/StandardFirmataEthernet.ino#L916) (but within the block), add this code:
```c++
if (counter++ >= 526) {
  DEBUG_PRINT("connected: ");
  DEBUG_PRINTLN(counter);
  counter = 0;
}
```

Compile, upload and open the Serial Monitor in the Arduino IDE (with your board connected via USB Serial) setting it to 9600 baud. You should see "connected: <current count>" printed every 10 seconds. If after a few minutes your application crashes, see if "connected" is still being printed. Then report back to me.

diegogarciaarriaza, 2017-08-08:
Thank you very much!, This afternoon I will try and then tell you something
Regards

alamitina, 2017-09-01:
```javascript
var board = new five.Board({port: "COM3"});
```

Just got a new Arduino and was getting that same error, but changing my board variable to the COM3 seems to work like a charm!

coolhandfluke, 2018-03-09:
I am getting similar issue with windows 7. Just tried to run led-blink.js code with windows7, uploaded standardfirmata that comes with arduino IDE. Below is the error log, any solution for this?

node led-blink.js
1520575717494 Connected COM3
1520575717511 Error  �`☻
(node:13776) UnhandledPromiseRejectionWarning: Unhandled promise rejection (reje
ction id: 1): Error: Uncaught, unspecified "error" event. ([object Object])
1520575727498 Device or Firmware Error A timeout occurred while connecting to th
e Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooti
ng

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the b
oard, wait approximately 11 seconds for complete reset, then run your program ag
ain.
events.js:165
      throw err;
      ^

Error: Uncaught, unspecified "error" event. ([object Object])
    at Board.emit (events.js:163:17)
    at Board.log (C:\psnl\learning\arduino\bossabledemo\johnny-five\lib\board.js
:630:8)
    at Board.(anonymous function) [as error] (C:\psnl\learning\arduino\bossabled
emo\johnny-five\lib\board.js:641:14)
    at Board.<anonymous> (C:\psnl\learning\arduino\bossabledemo\johnny-five\lib\
board.js:385:14)
    at ontimeout (timers.js:365:14)
    at tryOnTimeout (timers.js:237:5)
    at Timer.listOnTimeout (timers.js:207:5)

AndreyMartinez, 2018-03-19:
https://www.youtube.com/watch?v=yxl3D5hjWAs 

this is a video in which solve your problem is totally recommended

ArielPrez, 2018-09-07:
I had this issue. If you're using it through wifi it might be possible you're connected to the wrong router or connection.

diegogarciaarriaza, 2018-10-20:
Hi again. My last post was in 2017...
Sorry for that.

I just re-try connect with my arduino board via ethernet (firmataethernet) but I am having bad luck.
I am using the same code but now I don't get connect.

I have ping with my board and If I connect the board by USB, I have this exit in the console:

> connecting...
Connection failed. Attempting to reconnect...
connected
Connection failed. Attempting to reconnect...
Connection failed. Attempting to reconnect...
connected
Connection failed. Attempting to reconnect...
Connection failed. Attempting to reconnect...
connected
Connection failed. Attempting to reconnect...
>

All time is the same.

2 hours before, I get the problem installing johnny-five when tried to use serialport@4.0.7 but now I have downgraded to node v6 and that problem is not shown.
Now simply dont connect getting this:

> Please check that you've properly flashed the board with the correct firmware.
3|test2          | See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting
3|test2          | If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
3|test2          | Error: Uncaught, unspecified "error" event. ([object Object])
3|test2          |     at Board.emit (events.js:163:17)
3|test2          |     at Board.log (/var/www/domohouse_events/node_modules/johnny-five/lib/board.js:635:8)
3|test2          |     at Board.(anonymous function) [as error] (/var/www/domohouse_events/node_modules/johnny-five/lib/board.js:646:14)
3|test2          |     at Board.<anonymous> (/var/www/domohouse_events/node_modules/johnny-five/lib/board.js:390:14)
3|test2          |     at ontimeout (timers.js:386:11)
3|test2          |     at tryOnTimeout (timers.js:250:5)
3|test2          |     at Timer.listOnTimeout (timers.js:214:5)
>

Honestly I dont know what to do...

Any help?, thank you!

dtex, 2018-10-21:
Thanks for reporting back! Someone else will surely have the same issue and will find your solution.

closed 2018-10-21 by dtex

legoman849, 2020-12-12:
mine is saying this when i try to upload to my arduino uno im using node js

legoman849, 2020-12-12:
1607790942364 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
node:events:367
    throw err; // Unhandled 'error' event
    ^

Error [ERR_UNHANDLED_ERROR]: Unhandled error. ({
  type: 'error',
  timestamp: 1607790942364,
  class: 'Device or Firmware Error',
  message: 'A timeout occurred while connecting to the Board. \n' +
    '\n' +
    "Please check that you've properly flashed the board with the correct firmware.\n" +
    'See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n' +
    '\n' +
    "If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.",
  data: null
})
    at new NodeError (node:internal/errors:278:15)
    at Board.emit (node:events:365:17)
    at Board.EventEmitter.emit (node:domain:470:12)
    at Board.log (C:\Users\benal\node_modules\johnny-five\lib\board.js:642:8)
    at Board.<computed> [as error] (C:\Users\benal\node_modules\johnny-five\lib\board.js:653:14)
    at Timeout._onTimeout (C:\Users\benal\node_modules\johnny-five\lib\board.js:384:14)
    at listOnTimeout (node:internal/timers:556:17)
    at processTimers (node:internal/timers:499:7) {
  code: 'ERR_UNHANDLED_ERROR',
  context: {
    type: 'error',
    timestamp: 1607790942364,
    class: 'Device or Firmware Error',
    message: 'A timeout occurred while connecting to the Board. \n' +
      '\n' +
      "Please check that you've properly flashed the board with the correct firmware.\n" +
      'See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n' +
      '\n' +
      "If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.",
    data: null
  }
}

legoman849, 2020-12-12:
here is my code
var five = require('johnny-five');
var board = new five.Board();
  
board.on('ready', function(){
    var heart = [
    "01100110",
    "10011001",
    "10000001",
    "10000001",
    "01000010",
    "00100100",
    "00011000",
    "00000000"
  ];

  matrix = new five.Led.Matrix({
    addresses: [0x70],
    controller: "HT16K33",
    rotation: 1});
  
  matrix.clear();
  matrix.draw(heart);

});

calabiyauman, 2021-10-05:
Had this same issue and resolved it by specifying the port number with `board = new five.Board({ port: "COM6" });`