rwaldron/johnny-five #1262: Collision with Sense-Hat Components

yakeer, 2016-12-19:
Hello,
I'm using a Raspberry Pi 3, with a Sense Hat, and with a breadboard with a buzzer.

I'm controlling the Led Matrix with an npm called "sense-hat-led",
which I think is using ioctl as io.

Whenever I init a Board Class,
The LED Matrix is not longer responding to anything.

A reboot must be done in order to get it working again.
I'm initiating the board class with (nodeJS):

var raspi  = require('raspi-io');
var five = require("johnny-five");
var board = five.Board({
  io: new raspi()
});

as far as I understood, this is how it should be initiated in Raspberry Pi.

What am I doing wrong? or is it a bug/collision? how can I overcome this?

Thanks,
You're doing an awesome work!

dtex, 2016-12-19:
@yakeer raspi-io now allows you to specify which pins you want to initialize with Johnny-Five so you could tell it to leave your LED Matrix pins alone.

[This is the thread](https://github.com/nebrius/raspi-io/issues/76#issuecomment-262446648) where @nebrius announced it, and [here is the documentation](https://github.com/nebrius/raspi-io#api).

yakeer, 2016-12-19:
Thanks for the quick reply. 
Do you happen to know what pin is used for the led matrix?

dtex, 2016-12-19:
Honestly, I don't have a clue. Looking at the sense-hat-led I'm not sure that library will share the GPIO port. Maybe, rather than telling it to leave the LED Matrix pins alone you could use the ```includePins``` property so that you only initialize the pins you actually need in Johnny-Five. I'm not sure which pin(s) if any can be safely initialized without causing the Sense Hat to not work.

yakeer, 2016-12-20:
excludePins declared in Sense Hat did not work.
But includePins for the relevant ones did.

Thanks @dtex .

dtex, 2016-12-20:
Excellent

closed 2016-12-20 by dtex

rwaldron, 2016-12-20:
@nebrius @dtex duuuuudes that's really awesome

nebrius, 2016-12-20:
@rwaldron :D

it's funny, it's one of those features I never really knew we needed until I implemented it. I've used it in like two personal projects since, and it's only been a month!