rwaldron/johnny-five #1263: Johnny Five in Windows IoT

sarkarstanmoy, 2016-12-24:
I have connected arduino uno to Raspberry Pi 2 having Windows IoT preview edition installed. When I am  deploying  Univeral WIndows APP for Johnny Five from VS 2015 I am getting below error :

    console.log("It looks like serialport didn't compile properly. This is a common problem and its fix is well documented here https://github.com/voodootikigod/node-serialport#to-install");
    throw "Missing serialport dependency";

Can you please help me to understand why I am getting this error? Also, does windows IoT supports serialport

lyzadanger, 2017-01-25:
Hi @sarkarstanmoy 

Following the link in the error message itself brings me to the README for `node-serialport` (which as you can tell from the rest of the error message didn't compile right). There are some explanations/instructions in [this section of the README](https://github.com/EmergingTechnologyAdvisors/node-serialport#windows). Can you let me know if you were able to work through that section and/or your problem?

dtex, 2018-01-16:
Hi @sarkarstanmoy 

I'm going to close this since we never heard back. Feel free to re-open.

closed 2018-01-16 by dtex